rpng/open_vins #100: Ground-truth quaternion format

jamesdi1993, 2020-09-14:
Hi, 

I'm trying to use OpenVINs on a public dataset, and I have some confusion on the ground-truth format used by the simulator. Based on the example the format is:
~~~
timestamp tx ty tz qx qy qz qw Ptx Pty Ptz Prx Pry Prz
~~~

My question is the quaternion used here, is it the orientation from body to world or world to body? 

My confusion stems from this line: https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L47, and the feed_trajectory function is called here https://github.com/rpng/open_vins/blob/master/ov_msckf/src/sim/Simulator.cpp#L57 to load the ground-truth trajectory. 

It looks like (tx, ty, tz) is from body to world, i.e. the robot's position in the world frame, but (qx, qy, qz, qw) is from world to body, as there's a transpose there. Could you please confirm?

jamesdi1993, 2020-09-14:
Actually, I just realized that there's a comment explaining it here: 
https://github.com/rpng/open_vins/blob/master/ov_core/src/sim/BsplineSE3.cpp#L43

Sorry for the confusion!

closed 2020-09-14 by jamesdi1993

goldbattle, 2020-09-15:
We use JPL quaternions, so we actually read in the Hamilton q_ItoG from file which would be normally used by others. Thus we actually read in the transpose of the orientation that is in the file if it is a Hamilton. Specifically if we used a different [`quat_2_Rot()`](https://docs.openvins.com/namespaceov__core.html#adfb06397034cc6b346efb9517ed3757e) function converts from JPL to a SO(3) matrix.

We also worry about that transformation when we visualize the orientation in RVIZ (which uses Hamilton).
https://github.com/rpng/open_vins/blob/master/ov_msckf/src/core/RosVisualizer.cpp#L340-L342
```
// Publish our transform on TF
// NOTE: since we use JPL we have an implicit conversion to Hamilton when we publish
// NOTE: a rotation from GtoI in JPL has the same xyzw as a ItoG Hamilton rotation
```

TLDR: if you are using a JPL quaternion, yes it is q_GtoI, but if you are using Hamilton then the file will have q_ItoG.

jamesdi1993, 2020-09-15:
Thanks for your reply. That clears up things for me. 
~~~
// NOTE: a rotation from GtoI in JPL has the same xyzw as a ItoG Hamilton 
~~~
Is there any reference I can read about this note? I struggled to do the conversion myself and would like to understand it more.

goldbattle, 2020-09-20:
This issue has a bit of details: https://github.com/rpng/open_vins/issues/54#issuecomment-618457003
But in general you can look at the following two reports:
https://arxiv.org/pdf/1711.02508.pdf
http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf

[image]