rpng/open_vins #103: Landmark Representation Documentation

mhuzai, 2020-09-22:
Is there any documentation on what each of the landmarks defined [here](https://github.com/rpng/open_vins/blob/master/ov_core/src/types/LandmarkRepresentation.h#L39) mean? More specifically, how should one determine which representation to use for a given use case?

Thank you!

WoosikLee2510, 2020-09-22:
you can refer here: https://docs.openvins.com/update-feat.html

goldbattle, 2020-09-22:
Typically anchored has better accuracy, with the different inverse representations having minimal performance gains over the others. I would normally use inverse representations in cases were features are far away, but they come with some extra cost since they are required to be anchored. If you need speed then single inverse depth allows for more features but with reduced accuracy due to not estimating the bearing of the feature. The point representation part of the website should have a bit more details in terms of the mathematical definitions: https://docs.openvins.com/update-feat.html#feat-rep

mhuzai, 2020-09-25:
Thank you, this is really helpful. I have a few follow up questions:

1. What is "far away", roughly speaking? Is it several meters, like far away objects in a room, or tens of meters, like far away objects outdoors?
2. Based on what you said above, I would have thought that `ANCHORED_3D` would be the most common. But I noticed that `ANCHORED_FULL_INVERSE_DEPTH` is the most common in your [launch files](https://github.com/rpng/open_vins/tree/master/ov_msckf/launch). Why is that?
3. Global is never used in the above launch files. When would it ever be useful?
4. Finally, could you elaborate a little bit more on the difference between full and MSCKF depth?

goldbattle, 2020-09-28:
1. Depends how much the camera moves, but typically > 3 meters could be considered "far"
2. This is pretty arbitrary, you can try comparing the methods in simulation.
3. If you want speed global will be slightly faster compared to the anchored representations.
4. Single depth only estimates the and not the bearing of the feature. Thus the state is 1-dof instead of the traditional 3-dof.

mhuzai, 2020-09-29:
Thank you for all the answers!

closed 2020-09-29 by mhuzai