rpng/open_vins #106: Question about perform_anchor_change

zhuqingzhang, 2020-10-07:
Hello, sir, 
       I'm confused about Anchor change Jacobian and its propagation in **UpdaterSLAM::perform_anchor_change**. I don't know how to derive it. Are there any references about it?
Thanks a lot!

zhuqingzhang, 2020-10-07:
According to my understanding, 
First of all, I could get the following function: 

{}^G_p_f=func1(x_old,{}^{C_old}p_f)=func2(x_new,{}^{C_new}p_f)

so that I could get the function of {}^{C_new}p_f (i.e feature position in new anchor frame) with respect to x_old (old anchor frame pose), x_new (new anchor frame pose), and {}^{C_old}p_f (feature position in old anchor frame)
so, I have the function: 

 {}^{C_new}p_f =func(x_old,x_new, {}^{C_old}p_f )

from above function, I could get:

pf_new_error={\partial}func / {\partial}{x_old} *x_olderror+{\partial}func / {\partial}{x_new}*x_newerror+{\partial}func / {\partial}{p_f_old} * pf_olderror
                     =Hxold * x_olderror+ Hxnew * x_newerror + Hfold * pf_olderror
 However, from above derivation, I think it is different from the comments in the code [https://github.com/rpng/open_vins/blob/master/ov_msckf/src/update/UpdaterSLAM.cpp](url) see line #587 (// pf_new_error = Hfnew^{-1}*(Hfold*pf_olderror+Hxold*x_olderror-Hxnew*x_newerror)).

Could you please help me with some explanation ?
Thanks!

WoosikLee2510, 2020-10-07:
The eq [link](https://github.com/rpng/open_vins/blob/f56cac358509226074f5fbe0d956001464006d34/ov_msckf/src/update/UpdaterSLAM.cpp#L587) is solving unc1(x_old,{}^{C_old}p_f)=func2(x_new,{}^{C_new}p_f) :)

goldbattle, 2020-10-07:
There is a documentation page on this, let me know if it addresses your question:
https://docs.openvins.com/update-delay.html

zhuqingzhang, 2020-10-07:
Thanks for answers.
According to my understanding,  Hxnew, Hxold, and Hfold in equation [https://github.com/rpng/open_vins/blob/f56cac358509226074f5fbe0d956001464006d34/ov_msckf/src/update/UpdaterSLAM.cpp#L587](url)  is drived from equations in documentation page [http://docs.openvins.com/update-feat.html#feat-rep-anchor-xyz](url)
I know that
 **{}^G_p_f_error_old= Hxold*x_olderror+Hfold*pf_olderror,** 
and 
 **{}^G_p_f_error_new= Hxnew*x_newerror+Hfnew*pf_newerror.**
if  
**{}^G_p_f_error_old= {}^G_p_f_error_new ,**
then we have 
**pf_new_error = Hfnew^{-1}*(Hfold*pf_olderror+Hxold*x_olderror-Hxnew*x_newerror)**
just as the equation in the comments of the code.
However, why **{}^G_p_f_error_old= {}^G_p_f_error_new**? I think they should be a little different, as they are derived from different expansion point.
Is there any problem with my understanding?

WoosikLee2510, 2020-10-07:
{}^G_p_f=func1(x_old,{}^{C_old}p_f)
{}^G_p_f=func2(x_new,{}^{C_new}p_f)
**The position and its error** of the feature in global should be the same for both representations. If not, you are representing different feature.

goldbattle, 2020-10-07:
We are chaining through the global feature position.
As Woosik said above, we have the following:
```
Hxnew*x_newerror + Hfnew*pf_new_error = Hfold*pf_olderror+Hxold*x_olderror
```
Where we are taking the derivative of the global feature in respect to the anchored feature.
This comes from the taylor series expansion of (which by their formulation are exactly equal to each other):
```
f(x_old, p_FinA_old) = f1
f(x_new, p_FinA_new) = f2
f2 = f1
 f2_hat + H_f2 * x_err_f1 = f1_hat + H_f1 * x_err_f1
H_f2 * x_err_f1 = H_f1 * x_err_f1
H_x_new * x_err_x_new + H_f_new * x_err_p_FinA_new = H_x_old * x_err_x_old + H_f_old * x_err_p_FinA_old
x_err_p_FinA_new = H_f_new^{-1} ( H_x_old * x_err_x_old + H_f_old * x_err_p_FinA_old - H_x_new * x_err_x_new )
```

Another way to think about this is what if you: (1) convert p_FinA_old to p_FinG (2) convert p_FinG to p_FinA_new.

zhuqingzhang, 2020-10-08:
Thanks  alot for answers~
My original question was exactly why **f1_hat=f2_hat** (in the answer of goldbattle). I know that f1=f2 as they are **true value** of the same feature global position. However, **f1_hat=f(x_old_hat,p_FinA_old_hat)** and **f2_hat=f(x_new_hat,p_FinA_new_hat)** are the **estimated value**, they should be a little different.
But now I got it.  As goldbattle said, p_FinA_new_hat is actually from  (1) convert p_FinA_old_hat to p_FinG_hat(i.e. f1_hat) (2) convert p_FinG_hat to p_FinA_new_hat.  So that f1_hat==f2_hat. 
I think my understanding is correct now?

goldbattle, 2020-10-08:
They are going to be the same since we directly compute it using a frame transformation.
The mean of the state won't change at all during this since there is no update to consider.
```
// Calculate transform between the old anchor and new one
Eigen::Matrix<double,3,3> R_OLDtoNEW = R_GtoNEW*R_GtoOLD.transpose();
Eigen::Matrix<double,3,1> p_OLDinNEW = R_GtoNEW*(p_OLDinG-p_NEWinG);
new_feat.p_FinA = R_OLDtoNEW*landmark->get_xyz(false)+p_OLDinNEW;
```
[This](https://github.com/rpng/open_vins/blob/f56cac358509226074f5fbe0d956001464006d34/ov_msckf/src/update/UpdaterSLAM.cpp#L523-L526) takes the old feature (landmark), gets it anchor position, and then rotates / translates it into the new anchor frame. This skips the transformation to global and does it in one go, but enforces that globals are equal. Remember that `x_old_hat=x_new_hat` as we are only changing the feature state.

zhuqingzhang, 2020-10-08:
I got it!  Very appreciate! :)

closed 2020-10-08 by goldbattle