rpng/open_vins #107: Question : MSCKF features in update

lmuffang, 2020-10-12:
Hi!

I am working with the monocular version of open_vins with a fisheye camera with the right calibration parameters. 

I don't understand why I get `MSCKF update (0 features)` most of the time (sometimes I get 1 or 2 features in the update but never more). It seems that most of the features with more than 3 measurements fail to pass the triangulation. Hence they are erased from the `feature_vec`.

However, I still get a good amount of SLAM features (around 20 with a max of 75) when looking at the same scene as it should be. 

Is that a normal behavior or should there be more features for the MSCKF update?
I understood the concept of SLAM features but I don't think I quite understood how MSCKF features are used..

Thanks for the explanation!

goldbattle, 2020-10-13:
This depends on how many features are being included as SLAM features as these will be prioritizes over creating MSCKF features. If you track more features you should get more MSCKF tracks. It is ok if there are zero, just not that it is always zero. How many features are you tracking? 75 SLAM features is pretty large but you can do that if the computational budget is available.

lmuffang, 2020-10-14:
Thanks for your help!

I had 200 features max (`num_pts`) and 75 SLAM features (`max_slam`). I had only 1 or 2 MSCKF features occasionally, the rest of the time it showed 0 features in the MSCKF update and I always had around 20 SLAM features as long as a scene is seen for long enough.
By increasing the `num_pts` and setting less SLAM features, I get more MSCK features. Then yes the SLAM features were being prioritized as you said.

I just wonder why the number of features does not top off the maximum fixed by `num_pts`. Even with `num_pts` at 800 and 100 for `max_slam` , I get at most 25 SLAM features and around 15 MSCKF features (I have the computational resources on my computer as it does the filter update each 20ms).

goldbattle, 2020-10-15:
This likely sounds like the chi2 is rejecting values. Either your IMU noises are over confident or the image pixel noise is. You can try to increasing the chi2 multiplier to try to allow more through, but from what you explained the filter thinks the IMU provides amazing accuracy.

This could also occur in the case that your extrinsics and intrinsics are bad, but if you are estimating them only (you have calibration enabled?) then this should converge quickly from reasonable initial guesses.

lmuffang, 2020-10-23:
Indeed, by increasing the chi2 multiplier to 15 I get more reasonable number of feature in both updates.
Increasing the IMU noises (with chi2 multiplier at default or at 15) did not change much effect on the update.

I did not enable calibration, I tested it and most of the time it converged to values around the one I provided. I am also confident on my calibration, thus I turned off the online calibration in the algorithm.

Thanks for your help!

closed 2020-10-23 by lmuffang