rpng/open_vins #108: Update Types

mhuzai, 2020-10-17:
Could you explain what the difference is between an MSCKF update, a SLAM update, and a delayed SLAM init (I am not a SLAM person)? I was looking at the execution time per frame with Vicon Room Medium and noticed that MSCKF update is usually the fastest, then SLAM update, and then delayed SLAM init is the most expensive (almost 2x the other two). I understand that OpenVINS is VIO-only at the moment, so what does "SLAM" mean here?

Thank you!

goldbattle, 2020-10-21:
SLAM in this context means a feature that remains in the state vector for as long as it is seen/tracked continuously. After it is not seen any more it is removed from the state vector. MSCKF feature updates are all the "left over" features which are not inserted into the state vector as in the original MSCKF paper. Inserting SLAM features into the state vector (what we call delayed initialization) will take quite a bit a time as compared to the MSCKF update and SLAM update for features already in the state vector.

mhuzai, 2020-10-21:
Why is inserting SLAM features into the state vector expensive? And other than going through the code, how can I learn what the differences are between SLAM updates and MSCKF updates and how the two contribute to the pose?

goldbattle, 2020-10-22:
There are some details here about the initialization:
https://docs.openvins.com/update-delay.html

We touch on this a bit in Section III.B in the paper:
http://udel.edu/~pgeneva/downloads/papers/c10.pdf

But I recommend you read some related literature:
https://intra.ece.ucr.edu/~mourikis/papers/Li2012IROS.pdf
http://www.roboticsproceedings.org/rss08/p31.pdf

mhuzai, 2020-10-23:
Thank you for all the resources!

closed 2020-10-23 by mhuzai