rpng/open_vins #109: OpenVINS 1.0 vs. 2.0 Timing

mhuzai, 2020-10-20:
Hello,

I was looking at the timing data that is printed after each frame and observed significant differences between 1.0 (from last fall) and 2.0 (circa this April). The biggest difference is that tracking time has more than halved, and backend time has almost doubled. The decrease in tracking time is actually quite significant, resulting in a net 16% reduction in total time per frame. The tracker:backend time ratio was in the neighborhood of 70:30 in v1.0 but is around 40:60 in v2.0.

Do you have any idea why this may be? I saw that stereo KLT is no longer performed, but that only accounts for a third of the tracker's time. And I don't know what has changed in the backend (this also relates to #108).

mhuzai, 2020-10-20:
My apologies. There was a bug in our parameter setting code and we were not performing an apples to apples comparison of OpenVINS 1.0 and 2.0. With the same parameters, the only difference is that of stereo KLT, which is understandable.

But now I'm wondering if you could answer another question (instead of opening a new issue, might as well ask here): what are reasonable parameter values? I am not a SLAM researcher and only have a vague understanding of what all the parameters do. The code comments help, but if there is other documentation that I could look at, that would be great.
The two main parameters in terms of performance tuning seem to be number of points to track and number of SLAM/MSCKF features. What are good ranges for them?

Thank you!

goldbattle, 2020-10-21:
The majority of the time we found that tracking around 200 features, with 50 being SLAM / in the state seems to work well. Typically one wouldn't track more since it becomes more expensive, but the more features the better accuracy with diminishing returns up to a certain point.

mhuzai, 2020-10-21:
Thank you, this is really helpful. One more question: there is also a parameter called `max_slam_in_update`. What does that dictate and does it also have reasonable ranges that you could suggest.

goldbattle, 2020-10-22:
This parameter basically sets how many features should be "batched" into a single update. For example instead of doing an update of 50 features, you can do two of 25 each and be much faster. You can play with the number, but it only matter if you are using a lot of SLAM features and we typically set this to 25.

mhuzai, 2020-10-23:
Thanks again!

closed 2020-10-23 by mhuzai