rpng/open_vins #111: ［run_serial_msckf］process died when running the launch files in ov_msckf

leo-wkd, 2020-10-28:
Thanks for your excellent work！

When I tried to run tutorial.launch as you post on the website，I got the information below
**[run_serial_msckf-2] process has died [pid 8669, exit code -11, cmd /home/crz/ros/catkin_ws_ov/devel/lib/ov_msckf/run_serial_msckf __name:=run_serial_msckf __log:=/home/crz/.ros/log/741882fc-f4e0-11ea-b586-d36c088cb24a/run_serial_msckf-2.log].log file: /home/crz/.ros/log/741882fc-f4e0-11ea-b586-d36c088cb24a/run_serial_msckf-2*.log**

I installed all the dependency and compiled the the source code and I am sure that there is no problem with my ros version(kinetic). I could not figure out what problems I have encountered.

leo-wkd, 2020-10-28:
More information here.
Ubuntu16.04 with ROS Kinetic
Python2.7 Opencv3.4.5 Boost1.54.0

Everything is perfect when I catkin build the source code.

WoosikLee2510, 2020-10-28:
There is very little information in that error message. Can you try to figure out where the program is stopping at?

leo-wkd, 2020-10-28:
@WoosikLee2510 Thank you for your reply. In fact, I have figured out the problem is on my boost library. Two boost editions existed in my pc, which caused some conflictions when I ran the program. After reinstalling boost library, everything went well.
What surprised me is that the conflictions didn't occur when I built the source code, but occured when I tried to run the program.

closed 2020-10-28 by leo-wkd