rpng/open_vins #112: Rotation Error State Derivation

SubMishMar, 2020-10-30:
Hi, Thanks for such nice documentation of important derivations related to IMUs. I just have some questions about the rotation error state derivation. 
Specifically the following: 
[image]
Can you refer me to some reading or sources on this, because I am unable to reach at the same results as you have.

goldbattle, 2020-10-30:
Sola's work might be helpful (section 4.3 and 4.4): 
http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf

Note that they should be using a different rotation convention, what was is important is that throughout the codebase we use the same error state. In this case we model the error as being on the left side of a global to local rotation with a negative error. 
[image]
Additionally, we use quaternions in the codebase / state vector, but here it is easier to perform the derivation in SO(3) notation which is equivalent to its quaternion counterpart.

SubMishMar, 2020-10-30:
Okay, I will check it out and get back to you. Thanks. 

Also, how did you arrive at:
[image]

As far as I understand:
[image]

Why there is a -ve sign?

WoosikLee2510, 2020-10-30:
No, the other way.
w is the god-knows truth,
w with hat is the hat is the measurement (or estimated value).
You are subtracting true-measurement, thus your first eq has the minus sign

SubMishMar, 2020-10-30:
Oh! Ok. 

I am wondering why I thought the other way. I always thought the hat operator is not the measurement, probably because of the following:
[image]
[image]

But here hat refers to expectation (expected value) of the truth.

WoosikLee2510, 2020-10-30:
O- wait, I made a mistake.

w_hat is what you get from w_measurement and estimated bias b_hat, thus it is the estimated w so we use w_hat.

WoosikLee2510, 2020-10-30:
Let me clean this mess.

From https://docs.openvins.com/propagation.html,

w_m = w_true + b_true + n.
<-> w_true = w_m - b_true - n
you get the estimated w_hat by using estimated bias b_hat:

w_hat = w_m - b_est

thus the error of the estimated w, w_tilda is:

w_tilde = w_true - w_hat = w_m - b_true - n - w_m + b_est
 = - b_true + b_est  - n (we define error always true - est, so b_tilde = b_true - b_est)
 = - b_tilde - n

SubMishMar, 2020-10-30:
[image]

Yes. Even I derived the same. Basically, the terms with \tilde are errors with \hat are estimations of the god knows unknown truth and the terms without any symbols on them are the god knows unknown truths.

WoosikLee2510, 2020-10-30:
correct!

SubMishMar, 2020-10-30:
Correct me if I am wrong. In the documentation provided here: https://docs.openvins.com/propagation.html

We have (?)

[image]

Is there a special motivation/interpretation behind setting,
[image]

SubMishMar, 2020-10-30:
I think I understood the notation. This is how:
[image]
Interpreted it in 2D for simplicity.

SubMishMar, 2020-10-31:
Although I got a hang of the rotation equations in the EKF, I now have some problems with the position and the velocity parts. Especially with the following:
[image]

I am unable to derive the same. The result I obtain is below:
[image]

goldbattle, 2020-11-02:
In general it is not wise to try to subtract two rotation matrices. You might want to try leaving it as (R_true - R_hat)*g here. I believe in our documentation we have already pulled out the gravity term since it is independent to orientation change since it is in the global frame. If you want to go this route, you can try to substitute in and see if you can get the cancellations to remove the rotation in front of gravity.

SubMishMar, 2020-11-02:
I am not really subtracting the way one subtracts A-B. I have rather used the Taylor series expansion of R_true around R_hat.  Let my try what you have suggested in my derivations... I may want to estimate g as well as done in the tech report by Joan Sola which you referred me to earlier..

SubMishMar, 2020-11-02:
Is there any special motivation behind doing a left hand approximation of rotational error?

goldbattle, 2020-11-02:
If you want to estimate g, you would directly perturb and get a Jacobian of -I_3*dt for velocity for example:
[image]
Be warned that estimating gravity makes the system more unobservable (removes the roll and pitch observability).

Any form can be picked (for example right hand error is good for observability reasons as in the MSCKF 2.0 paper, which we address through FEJ), but this was picked purely based on convention with existing MSCKF literature.

SubMishMar, 2020-11-05:
I got what you meant by pulling out the gravity terms and that makes the derivation easy..
[image]

One need not define \delta a and can directly derive the error state mean propagation for velocity and reach at the same result that you have..

[image]

SubMishMar, 2020-11-08:
I am trying to dig deeper into the code. Especially the parts concerning State Propagation. You mention in the [documentation](https://docs.openvins.com/classov__msckf_1_1State.html) "This state has all the current estimates for the filter. This system is modeled after the MSCKF filter, thus we have a sliding window of clones." I am little confused by what is meant by clones here? I think you are talking about a window of state values between time t_{k-n} to time t_{k}, where n is the size of the sliding window. I may be wrong, so I want to double check.

goldbattle, 2020-11-08:
Cloning is the procedure of "cloning" the current pose such that after cloning there are two copies of the pose.
One will advance forward in time, while the other will remain at the historical time.

These related works should help (which should be cited from the original MSCKF paper if I remember correctly):
https://www-users.cs.umn.edu/~stergios/papers/sc-kf_icra02.pdf
https://www-users.cs.umn.edu/~stergios/papers/TRO-SC-KF.pdf

goldbattle, 2020-11-17:
Feel free to re-open when you have any further issues.

closed 2020-11-17 by goldbattle

SubMishMar, 2020-11-28:
This might be a naive question but I do not understand why do we need a jerk to initialize the VIO? Especially for stereo, where the scale is already known? 
Even for mono, won't two (or three) different frames with smooth motion between them suffice to initialize the VIO?

goldbattle, 2020-11-28:
The jerk is just to detect when the IMU starts moving and classify which segment is stationary if this is what you are referring to. Right now we don't have SFM initialization, so you need this stationary detection phase to perform our [stationary initialization procedure](https://docs.openvins.com/classov__core_1_1InertialInitializer.html). I believe I have answer this before, so I recommend searching the issues for "initialization (e.g. #58 and others have some details that might help you).

Open another issue if you have any further questions about initialization after looking through the past ones.

FengQiaojun, 2022-11-18:
I want to follow up on the question regarding the transition function of rotation error.
In this part
[image]
Do we use these tricks?

${}^{I_{k+1}}_{I_k}\mathbf{R} = \exp(-\boldsymbol{\omega}\Delta t)$   ->   Write the rotation matrix as the exp map of augular velocity
$\exp\left(\theta + \delta\theta \right) \approx \exp(\theta) \exp(\mathbf{J}_r(\theta)\delta\theta)$ for small $\delta\theta$
$\exp(-\delta\theta) \approx \mathbf{I} - \lfloor \delta\theta \rfloor$ for small $\delta\theta$

And should we convert from above to this
[image]
Originally it was an equation between two 3x3 matrices and the final result is an equation between two 3-d vectors. How is this achieved? I guess I miss some tricks of converting skew symmetric matrix/cross product.

Any hints or guidelines are appreciated!

FengQiaojun, 2022-11-23:
OK I got some idea on how the approximation between two matrices is converted to the approximation between two vectors. 
The main idea is to convert both sides to skew-symmetric matrices and then $\lfloor \boldsymbol{a} \times \rfloor = \lfloor \boldsymbol{b} \times \rfloor \Rightarrow \boldsymbol{a} = \boldsymbol{b}$ for $\boldsymbol{a}, \boldsymbol{b} \in \mathbb{R}^3$. 
Another property used is $\mathbf{R} \lfloor \boldsymbol{\theta} \times \rfloor \mathbf{R}^\top = \lfloor \mathbf{R} \boldsymbol{\theta} \times \rfloor$ for $\mathbf{R} \in SO(3)$.

goldbattle, 2022-11-23:
You can refer to this technical report for mathematical identities you are trying to define:
http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf

For the SO(3) ones, you I recommend Barfoot's book (specifically Section 7.1.5 and Equation 7.75):
http://asrl.utias.utoronto.ca/~tdb/bib/barfoot_ser17.pdf

In an old tech report I have some useful identities you can also check:
https://pgeneva.com/downloads/reports/tr_async.pdf
https://github.com/rpng/open_vins/files/10078528/useful_identities.pdf