rpng/open_vins #115: environmental landmarks in state vector?

suho0515, 2020-11-30:
hello sir, i got a short question about the state vector of ov_msckf.

i understood that ov_msckf is motivated by "A Multi-State Constraint Kalman Filter for Vision-aided Inertial Navigation".

and it said that the state vector is consist of imu poses/camera poses like picture below.
[image]
 
get the 3d feature position using equation below.
[image]

G_p_fj could be calculated using least-squares minimization.

but in OpenVINS paper, it says state vector include landmark X_M, G_p_fm like below.
[image]

my question is,
i thought that the advantage of msckf is there is no need to input 3d feature position to state vector.
so it can reduce complexity of computing and reduce memory using.

but in ov_msckf paper, it include 3d feature position in state vector.
does it has different meaning with original msckf paper?

and thanks all your brilliant work.
admire new update too :).

WoosikLee2510, 2020-11-30:
OpenVins is based on the msckf and that allows the estimator to run without storing landmarks in the state. Having landmarks in the state (slam features) is optional, which is often useful when the system can observe the same feature for a long period of time. So if you do not use slam features, the system follows the msckf idea, if use, it is a more general system that can handle both msckf way and slam way of treating the visual measurement.

goldbattle, 2020-11-30:
While the original MSCKF paper is a key part to enabling realtime visual-inertial estimation, there have been many works which have built off this to further improve performance and accuracy. The primary motivation of including features into the state vector is to improve accuracy and also handle MAV / vehicle stopping cases (where features are constantly re-observed). I recommend you take a look at Mingyang Li's works along with the other literature cited in the [paper](http://udel.edu/~pgeneva/downloads/papers/c10.pdf). For example:

Li, Mingyang, and Anastasios I. Mourikis. "High-precision, consistent EKF-based visual-inertial odometry." The International Journal of Robotics Research 32.6 (2013): 690-711. https://journals.sagepub.com/doi/abs/10.1177/0278364913481251

Li, Mingyang, and Anastasios I. Mourikis. "Optimization-based estimator design for vision-aided inertial navigation." Robotics: Science and Systems. Germany: Berlin, 2013. http://roboticsproceedings.org/rss08/p31.pdf

suho0515, 2020-11-30:
@WoosikLee2510 @goldbattle 
thank you very much! i totally understand what ov_msckf is trying to do, and thanks for the paper, i sure look at it :)
there are clear answer for this question, so i'll close this issue :)

closed 2020-11-30 by suho0515