rpng/open_vins #119: Rotational drift in gyro-only odometry is quite awful than I expected

jongwonjlee, 2020-12-04:
Thanks to the OpenVINS library, I was able to compare the accuracy of orientation estimation in monocular visual odometry and gyro-only odometry in simulation. For visual odometry, frame-by-frame relative rotations are accumulated along with time, whereas the state propagation formula for IMU, which can be seen in both the OpenVINS official webpage and the paper (and as `Propagator::predict_mean_discrete` member function in the directory `ov_msckf/src/state/Propagator.cpp`), was used to carry out gyro-only odometry. For a fair comparison, all noises or biases are assumed to be zero. So to speak, the gyroscope measurement can be assumed to be a ground-truth measurement for each timestep. The camera shutter speed was set to 10Hz whereas the gyroscope measures every 200Hz. [Here](https://www.dropbox.com/sh/7l7eysdc4drhxen/AADgVmSu-yX3Jh5xMbaf8_uYa?dl=0) are the results.
Surprisingly, the former performed better than the latter except for some points where frame-to-frame epipolar geometry fails. This outcome conflicts with my intuition that the former should estimate the orientation more accurately since it has a way more large sampling rate. The only point suspicious is the IMU propagation model itself, but it also was not the case since the accumulated drift error did not go down even when I increase the gyroscope's sampling rate. If the propagation model was problematic, the error should have decreased when I have don't so but it didn't in practice.
Is the IMU propagation model as shown in [tutorial](https://docs.openvins.com/propagation.html) not perfect mathematically? If it is, how can we improve it? Or, if not, what leads it to such a huge amount of drift error?

goldbattle, 2020-12-04:
While increasing the IMU sensing rate can help address the assumptions used during integration this does not reduce the noise of the inertial measurements, and under the case of noise one would expect inertial integration to quickly diverge. One of the key assumptions during the integration is that the IMU measurements are constant over the integration period, thus if you go to a higher sampling rate (1Hz vs 1kHz) this assumption should hold more.

I recommend you also try the [RK4 integration](https://docs.openvins.com/classov__msckf_1_1Propagator.html#a00d3d1e121e222bed030e2ed0584ab20) method. My intuition would think that under no noise and high enough IMU frequency the integration should have extremely small errors. If you are using the included simulator, you need to ensure that you set the IMU noise and walks to be **zero** (from a quick glance at your repo, I am not sure this is the case as you claim...).

I don't have any experience in this exact problem, but I recommend you searching the literature and looking at the works we cite. Specifically this tech report discusses the assumptions used during the orientation integration:
http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf

jongwonjlee, 2020-12-05:
Hmm... That's still suspicious. Following your recommendation, I tried RK4 integration (as in the function `Propagator::predict_mean_rk4`) as a quaternion propagation model with zero gyroscope noise, but the results make no difference even when I increased the gyroscope's sampling frequency up to 500Hz. Here, I attach results on ideal-gyroscope-based odometry for both indoor and outdoor environments - `udel_gore` and `udel_neighborhood`- with three different sampling rates - 10Hz, 100Hz, and 500Hz, respectively.

[image]
[image]

Especially, these outcomes conflict with my intuition since the drift should have gone down when I increased the sampling frequency. I couldn't find out what's actually the problem. Would you mind if you have a look at my repo? It's inside the file `ov_calib/src/quat_propagation.cpp` on my branch `jongwonjlee/calib-simulator`.

closed 2021-01-15 by goldbattle