rpng/open_vins #120: Measurement Jacobian Calculation

SubMishMar, 2020-12-04:
Hi,
I am going through you code where you assign the measurement Jacobians with respect to the state variables to matrices like H_f and H_x. According to the documentation, H_f is the Jacobian w.r.t. the feature error state variable and H_x is the Jacobian w.r.t. extra error state variables like anchored state and also calibration states. The variable x_order takes care of these extra variables. My question is where do you deal with the jacobian blocks for the current state of the IMU? 

Let us assume that in our scenario we have 3 kinds of state variables:
1. Anchored State: X_a (of type PoseJPL)
2.  Extrinsic Calibration: X_c (of type PoseJPL)
3. Current IMU: X_I (of type IMU)

I can in the code, especially here: https://docs.openvins.com/classov__msckf_1_1UpdaterHelper.html#a866dde286826115155ad40fb2066655c that you have dealt with 1. and 2. but where do you deal with 3? I think I am missing something and I will appreciate you kind help. Thanks.

goldbattle, 2020-12-04:
We clone directly after propagation, thus we write all our features as function of the `clones_IMU` PoseJPL objects. If we had a special case for the latest IMU pose, then we would have to had additional time offset logic (the clones are at the true camera time, while the propagated state is at the current best estimate for the time offset). Writing all measurements as a function of the cloned poses both simplifies the code and doesn't occur any extra computational cost.

The detailed equations are provided here on [the documentation website](https://docs.openvins.com/update-feat.html).
[image]

These last two Jacobians are in respect to whatever clone the state is observed from:
https://github.com/rpng/open_vins/blob/e3cc117a6170d1d570ad6ebfee7eda873ef4bede/ov_msckf/src/update/UpdaterHelper.cpp#L542-L543

The anchored state derivative of measurement in respect to its pose is here:
https://github.com/rpng/open_vins/blob/e3cc117a6170d1d570ad6ebfee7eda873ef4bede/ov_msckf/src/update/UpdaterHelper.cpp#L545-L549

SubMishMar, 2020-12-04:
Understood! Thank you so much. Yes, I had missed the StateHelper::augment_clone() function you have invoked from the Propagator::propagate_and_clone() function.

closed 2020-12-04 by SubMishMar