rpng/open_vins #121: catkin_make failing due to static_assert in Eigen

ShreyasManjunath, 2020-12-07:
Hi,
Thank you for the development.
I am trying to compile openvins on ubuntu 18.04 but encountering the following error. 

> /usr/local/include/eigen3/Eigen/src/Core/util/StaticAssert.h:33:40: error: static assertion failed: YOU_TRIED_CALLING_A_VECTOR_METHOD_ON_A_MATRIX
>      #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
>                                         ^
> /usr/local/include/eigen3/Eigen/src/Core/util/StaticAssert.h:33:40: note: in definition of macro ‘EIGEN_STATIC_ASSERT’
>      #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
>                                         ^~~~~~~~~~~~~
> /usr/local/include/eigen3/Eigen/src/Core/Dot.h:74:3: note: in expansion of macro ‘EIGEN_STATIC_ASSERT_VECTOR_ONLY’
>    EIGEN_STATIC_ASSERT_VECTOR_ONLY(Derived)
>    ^
> /usr/local/include/eigen3/Eigen/src/Core/util/StaticAssert.h:33:40: error: static assertion failed: YOU_TRIED_CALLING_A_VECTOR_METHOD_ON_A_MATRIX
>      #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
>                                         ^
> /usr/local/include/eigen3/Eigen/src/Core/util/StaticAssert.h:33:40: note: in definition of macro ‘EIGEN_STATIC_ASSERT’
>      #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
>                                         ^~~~~~~~~~~~~
> /usr/local/include/eigen3/Eigen/src/Core/Dot.h:75:3: note: in expansion of macro ‘EIGEN_STATIC_ASSERT_VECTOR_ONLY’
>    EIGEN_STATIC_ASSERT_VECTOR_ONLY(OtherDerived)

This error pops out due to a line in RosVisualizer.cpp in ov_msckf pkg.
Line 667
` double ori_nees = 2*quat_diff.block(0,0,3,1).dot(covariance.block(0,0,3,3).inverse()*2*quat_diff.block(0,0,3,1)); `

Seems like `Eigen::Matrix<double,4,1> quat_gt, quat_st, quat_diff;` quat_diff is indeed a Matrix.

Does anyone had similar experience?
Thank you

Shreyas

goldbattle, 2020-12-08:
What version of Eigen3 are you linking against?

ShreyasManjunath, 2020-12-08:
Hi,

I use Eigen3 3.3.9 (Newest). 

Regards,

Shreyas

closed 2020-12-11 by ShreyasManjunath

goldbattle, 2020-12-11:
Were you able to address this issue?

ShreyasManjunath, 2020-12-11:
Hi,

Yes. Seems like open_vins is not compiling for Eigen 3.3.9.
I switched to Eigen 3.3.4 and was able to compile it.

Btw. Do you have any separate branch for OpenCV 4 support?

Thank you once again.

Shreyas Manjunath

goldbattle, 2020-12-11:
You can try out #117 as it should support v4.
Post in that PR if you run into any issues using the newer opencv (it should default to 4 if you have 3 and 4).

hellovuong, 2021-06-30:
I fixed the error use dot() method for matrix quat_diff in Eigen 3.3.9, I change line [615](https://github.com/rpng/open_vins/blob/c6f7e947e82485873495123a1841bd57f8041a33/ov_msckf/src/core/RosVisualizer.cpp#L651), from:
`double ori_nees = 2 * quat_diff.block(0, 0, 3, 1).dot(covariance.block(0, 0, 3, 3).inverse() * 2 * quat_diff.block(0, 0, 3, 1));`
to:
`Eigen::Vector3d temp_vec = covariance.block(0, 0, 3, 3).inverse() * 2 * quat_diff.block(0, 0, 3, 1);`
`double ori_nees = 2 * quat_diff.adjoint()*temp_vec;`
By instruction about _Dot product_ in [Eigen page](https://eigen.tuxfamily.org/dox/group__TutorialMatrixArithmetic.html#title6)

goldbattle, 2022-02-10:
PR https://github.com/rpng/open_vins/pull/226 should address Eigen issue for newer versions.