rpng/open_vins #122: Jacobian calculation for rotation only measurements.

SubMishMar, 2020-12-07:
I went through your documentation and much of the theory related to calculation of Jacobians is associated to cases when the rotation matrix is multiplied to a vector (R v). I am particularly concerned with cases where we have sensors that provide rotation between time instants (in terms of a rotation matrix). I will be glad if you let me know if the equations and math I derive below look ok to you because I could not find any reference for this where all the details are provided at one place. 

Let us assume we have a sensor which provides the difference between two frames $\{i-1\}$ and $\{i\}$. Let's say the sensor provides it in terms of a rotation matrix $R^{i-1}_{i}$. Here the global frame of reference is $\{G\}$. So the measurement function can be written as:
[image]

the LHS is the measurement and the RHS is composed of rotational state variables $R^{i-1}_{G}$ and $R^{i}_{G}$. The goal is to calculate the Jacobian of this measurement function with respect to the aforementioned state variables. 

For this, the rotation matrix can be perturbed as follows:
[image]

Substituting and manipulating, one gets: 
[image]

Where: 
[image]

Is it correct to say the measurement Jacobians with respect to the rotation state variables are:
[image]

Appreciate your time and help :)

goldbattle, 2020-12-08:
You need to define what your LHS error state is.
If the error for the LHS is R=exp(-theta)*R_hat, then I believe your derivations should be fine.

SubMishMar, 2020-12-08:
Yes, that is what I define my error state as. Using the left multiplication with exp of a  -ve delta angle vector. Thanks for clarification.

goldbattle, 2020-12-08:
You can have different measurement error defined (how you compute your residual, your LHS), then what your state error states. Thus you need to carefully define both of them (i.e. left or right orientation error states with positive negative signs).

closed 2020-12-08 by goldbattle