rpng/open_vins #125: why not covariance square root kalman filter?

tadashi-hamada196, 2020-12-22:
Almost every msckf algorithm on github is implemented by using double floating point，and I know single floating point may bring numerical precision problem. Covariance square root kalman filter can solve the numerical precision problem, and coding with floating point can be 4 times fast in ARM platform because of ARM NEON, so why openVINS use it?

goldbattle, 2020-12-23:
While I do not have explicit timings for you, modern ARM systems don't suffer from that poor of performance that one would expect. Typically the issue with these systems (such as the nvidia jetson or raspberrypi) is the feature tracking frontend which we rely on OpenCV. There are many benefits and disadvantages to the different formulations (sqrt, inverse, covariance, etc).

tadashi-hamada196, 2020-12-23:
Thanks a lot

closed 2020-12-23 by tadashi-hamada196