rpng/open_vins #131: Use of very wide angle fisheye cameras

visocopter, 2021-01-14:
Hi,

is it possible to use a fisheye camera with a very high FOV (>180 deg) fisheye lens with Open Vins?

I have a global shutter camera and would like to use a circular 180 degree fisheye lens. 
As I understood the documentation, there are two camera models supported: _pinhole-equi_  and _pinhole-radtan_.

Would the pinhole-equi model sufficient for this task? 
Or generally: What would have to be done to use a circular 180 degree fisheye camera? 

Keep up the great work!

Cheers,
James

goldbattle, 2021-01-14:
We leverage the opencv fisheye mode:
https://docs.opencv.org/master/db/d58/group__calib3d__fisheye.html

My understanding that their model does not support very large FOV cameras. Additionally, I think some of the estimation code for triangulating features would need to be changed as right now we assert that features must lie in front of the camera, thus for >180deg this won't hold since features can lie behind the camera frame. I don't have any experience with these cameras, but I think another issue is performing feature tracking on these large FOV angles due to the large amounts of distortion. You can try selecting a region of interest in the middle of the image to reduce the FOV is you are stuck with a high FOV sensor.

visocopter, 2021-01-15:
Alright thank you very much. 
Would a 160 degree lens work then with the existing models since the image wouldn't be behind the frame?
(Like the Intel Realsense T265, which has a 160 degree FOV I think)

goldbattle, 2021-01-15:
Yes, you can checkout the other issues created for that.

- https://github.com/rpng/open_vins/issues/64#issuecomment-628898705
- https://github.com/rpng/open_vins/issues/97
- https://github.com/rpng/open_vins/issues?q=T265+

visocopter, 2021-01-15:
Okay thank you!

closed 2021-01-15 by visocopter