rpng/open_vins #133: Robustness to Camera and IMU Timing

mhuzai, 2021-01-18:
How robust is OpenVINS to variation in camera and IMU timing? Assume that new camera images arrive every 50 ms in general. Then, if a new image arrives 48 ms or 52 ms after the previous one, would the algorithm work fine? A more egregious case of this would be when an entire image gets dropped and 100 ms elapse between consecutive images. What would happen then? Same with the IMU readings.

Another question I have is about robustness to timestamping accuracy. What happens if two IMU readings arrive 2 ms apart but get timestamped as 2.05 ms apart?

In practice, we use a ZED Mini camera without any special timing, and that works fine, but I would just like to make sure.

Thank you!

goldbattle, 2021-01-19:
The actual time of arrival of the image shouldn't matter much, but the timestamp varying (typically called jitter) could be an issue. If the timestamp is accurate, then when it arrives it shouldn't really matter at besides the filter will be a bit delayed in updating. I don't have any experience with the jitter aspect, but I don't think small amounts (such as 0.05ms) should have any impact and is typically not something people worry about unless they are on very low power systems such as older consumer phones.

If the a camera image is dropped every once and a while that isn't really an issue. If an IMU is dropped, it is a lot worst case as now we now are integrating over a larger time period, but the system should be able to seamlessly handle this case. Drops would be more of a concern if they are happening frequently, or for long periods (greater than just a single reading).

mhuzai, 2021-01-20:
Thanks, that's super helpful.

One more question: (assuming no jitter here) what happens if the current time is t=30 and due to some delay the next camera frame that arrives is for t=10 and the available IMU readings are for t=30? I'm guessing since the state only gets updated on camera frames, we would just advance the state to t=10 and only use the IMU readings till that point?
Basically, if independent threads are publishing camera and IMU data and can get arbitrarily delayed, do things still work fine?

goldbattle, 2021-01-25:
Yes, that would be the case. Once the camera gives t=10, the imu would be used to propagate to t=30 for the odometry topic.
Of course this is not ideal, and is something that should be fixed on the hardware side if it is a 20second delay.

mhuzai, 2021-01-26:
Thanks again!

closed 2021-01-26 by mhuzai