rpng/open_vins #134: Code for Paper "Visual-Inertial-Wheel Odometry with Online Calibration"

adas-dev, 2021-01-22:
Dear OpenVINS authors,

A while ago you published a paper "Visual-Inertial-Wheel Odometry with Online Calibration" (VIWO) and you also have a video on YouTube illustrating how the system performs in different sensor configurations (https://www.youtube.com/watch?v=4DH-J5OtjLc).

I am wondering whether you plan to make the code used for this paper public, and, if the answer is "yes," what is the time estimate to see these changes to OpenVINS in the `master` branch?

goldbattle, 2021-01-25:
At this point we don't have any timeline on if / when this codebase will be open sourced.
Additionally, it would be released as a separate project as openvins' goal is to provide a visual-inertial baseline.

closed 2021-01-25 by goldbattle

duyanwei, 2025-09-03:
Hello, sorry to reopen this issue, I know it has been a long time. I just wanted to ask if you have released the wheel-aided system code elsewhere. Thank you for your time and help!