rpng/open_vins #135: Time offset Jacobian Calculation in MSCKF Update

SubMishMar, 2021-01-24:
Hi, 

I am trying to understand how you perform online time offset estimation. As written in a comment in the code I am referring to the paper titled: "Online temporal calibration for camera–IMU systems: Theory and algorithms ". I think I have understood the paper but I am unable to locate where in the code you have calculated the measurement Jacobians w.r.t the time offset state variable for the MSCKF update (and other update) step. Something akin to the last Equation in the set of Equations (26) in the aforementioned paper.
[image]

I was able to find the corresponding Jacobians for state propagation here: https://github.com/rpng/open_vins/blob/16f5c37f2f9a13f6641f160e2355a763315327ba/ov_msckf/src/state/StateHelper.cpp#L590-L600

But I am looking for corresponding Jacobians for state update. I will appreciate your help with this.

Thanks. 
Subodh.

goldbattle, 2021-01-25:
The update will not require a derivative in respect to time offset since the cloning procedure has cloned the state at the "true" time. As mentioned, the paper has two different methods: eq (26) if your clones are at arbitrary IMU times, and eq (37) if you have taking into account the time offset during the cloning procedure.
https://docs.openvins.com/classov__msckf_1_1StateHelper.html#ade3a86a89f15cc6e03a6430fd1dee694

From below eq (35):
> To account for these additional parameters, one can compute the Jacobians of the feature measurements with respect
to them, similarly to what was described in the previous
cases. However, in pose-based methods, an alternative, and
simpler, approach exists. Specifically, the only filter operation that needs to be changed, compared to the original methods, is state augmentation

SubMishMar, 2021-01-27:
Thanks! I am closing this. I will reopen if I have further questions.

SubMishMar, 2021-03-03:
@goldbattle just verifying my understanding of this: 

basically what you do here (and also what is conveyed in the paper titled: "Online temporal calibration for camera–IMU systems: Theory and algorithms ") is that you first propagate the IMU pose to t+td and then add that pose to the list of clones (state vector augmentation), at the same time you have to augment the covariance matrix which you do here: https://docs.openvins.com/classov__msckf_1_1StateHelper.html#ade3a86a89f15cc6e03a6430fd1dee694

As far as the MSCKF updates are considered you do not bother about the time offset td because the states wrt which the  Measurement Jacobians need to be calculated are basically IMU clones which have already taken into account the time offset variable td.

SubMishMar, 2021-03-03:
@goldbattle just verifying my understanding of this:

basically what you do here (and also what is conveyed in the paper titled: "Online temporal calibration for camera–IMU systems: Theory and algorithms ") is that you first propagate the IMU pose to t+td and then add that pose to the list of clones (state vector augmentation), at the same time you have to augment the covariance matrix which you do here: https://docs.openvins.com/classov__msckf_1_1StateHelper.html#ade3a86a89f15cc6e03a6430fd1dee694

As far as the MSCKF updates are considered you do not bother about the time offset td because the states wrt which the Measurement Jacobians need to be calculated are basically IMU clones which were estimated taking into account the time offset variable td.

Another question is, do you model the time offset as a constant or as a time-varying quantity?

goldbattle, 2021-03-08:
The cloning procedure itself is a function of the time offset since the propagated pose is at the estimate, while we want to clone a the true camera timestep. If you wanted to model the time offset, then you would inject noise during propagation (i.e. when you move the state forward in time, the time offset should become more uncertain like how biases are modeled). Right now we do not model as time varying since this inject is not performed in the propagator.

closed 2021-03-08 by goldbattle