rpng/open_vins #137: Running OpenVINS GPU-accelerated?

visocopter, 2021-01-27:
Hi,

I am playing with the thought of implementing a CUDA Version of OpenVINS. Before that, I wanted to ask if this is a good idea?

I basically want to use OpenVins with multiple cameras (around 4-6 as 2-3 stereo configurations) on a Jetson Xavier NX onboard computer in real-time. 
I am not sure, if the CPU can handle this by it's own. Also, I am running other calculations simultaneously on the CPU.
Would it make sense to run the core algorithms on a GPU instead of the CPU? How much would the performance actually increase?

Cheers,
James

goldbattle, 2021-01-31:
I have not personally played with implementing a GPU version of the underlying EKF, but this might have speedups if the system is large and likely with just a few clone states accelerating might not have any performance gains.

The most direct place to speedup the system is in the frontend feature tracking which takes the majority of the time on resource constrained systems. A while back I looked at using the OpenCV GPU accelerated *descriptor* matching but only saw a very marginal speed up x1.5 on a Jetson Nano. Additionally this was only the case when the amount of features were in the 2 thousands and I saw that actually on the resource constrained systems the OpenCV GPU accelerated descriptor extraction and matching hurt frontend performance. On a 1080ti, there was usually greater then x2 speedup even for 100s of features. I did not play with the OpenCV GPU KLT implementation but you could also investigate.

If you are interested in playing around with the GPU acceleration I recommend you checkout [this project](https://github.com/uzh-rpg/vilib). I have wanted to incorporate this as an optional frontend that can be used due to its really impressive extraction and tracking (< 3ms on average for both extraction and tracking every frame).
https://github.com/uzh-rpg/vilib

Also you should try the default repo on your platform and see if it works. We have run on a raspberry pi / nano in realtime stereo with around 100 features at 25-30fps with reasonable success. It is important that you pass the correct compile flags such that you compile the repo with vectorization for the underlying Eigen3 matrix operations.

closed 2021-02-04 by goldbattle

mzahana, 2023-06-24:
Hi @goldbattle 
Regarding your comment

>  It is important that you pass the correct compile flags such that you compile the repo with vectorization for the underlying Eigen3 matrix operations.

What flags to set to build this package on  Jetson Xavier NX?
I am using ROS 2.

Thanks

ssaigarimella, 2025-04-03:
@mzahana Was wondering the same, can you please share what flags you built it with on Xavier NX, if you were able to? Thanks