rpng/open_vins #138: How to derive the orientation propagation equation?

niebayes, 2021-02-07:
I checked the IMU error state progation [page](https://docs.openvins.com/propagation.html#error_prop), but I got stuck in the derivation of oritentation propagation equation. In particular, it's this line that causes the confusion. 
[image]
I know the JPL convention is adoped and hence the minus signs in front of the skew-symmetric matrices. And I know the **exp** operator is actually the "vectorized" **Exp** operator defined the "State Estimation for Robotics" which means it maps the rotation vector to the SO(3) Lie group. Also the composition of the nonimal state $$\hat{\omega}$$ and the error state $$\tilde{\omega}$$ gives the "true" state. 
But what places a minus sign in front of the omegas ? Due to local from/to global convertion or JPL convertion, or something else tricky?

SubMishMar, 2021-02-07:
This thread https://github.com/rpng/open_vins/issues/112 might help you.

goldbattle, 2021-02-08:
Does #112 address your issue?

niebayes, 2021-02-08:
@goldbattle Yes, it's helpful!

closed 2021-02-08 by niebayes