rpng/open_vins #140: Euroc timestamps mismatch

hoonkai, 2021-02-15:
Hi

I'm trying to work with V1_01_easy and I've noticed that the timestamps in the bag file do not quite match with those in the [V1_01_easy.csv](https://github.com/rpng/open_vins/blob/master/ov_data/euroc_mav/V1_01_easy.csv) file. For instance, the first frame in the bag file has timestamp `Bag Time: 1403715271.716820` but the first entry in the gt CSV has timestamp (ns) 1403715273262142976, which is about a 2 seconds difference. Does anyone know why this is the case? Wouldn't the discrepancy affect the evaluation?

Thanks

nmerrill67, 2021-02-16:
This is expected. Most of the sensor data (including ground truth) has timestamps that start slightly off of the others (sometimes even with nonzero time offsets from each other). This can be accounted for in multiple ways. You'll notice in the launch files that you can initialize with the groundtruth global frame, but this is not ideal since it can cause the IMU initialization to fail. The better route is aligning the estimated and GT trajectories, which may start at different times with different global reference frames, with an optimization problem. If you look at the [evaluation docs](https://docs.openvins.com/evaluation.html) you can see how to do it with open_vins.

goldbattle, 2021-02-16:
This csv should be generated using the [vicon2gt](https://github.com/rpng/vicon2gt) utility. It likely starts later than the bag start time since that is when the first vicon pose is received. You can checkout the original [V1_01_easy_original.csv](https://github.com/rpng/open_vins/blob/master/ov_data/euroc_mav/V1_01_easy_original.csv) to see if it also has the issue you have mentioned. The poses which occur before the time in the groundtruth should just be ignored and not included in the error calculation.

closed 2021-02-21 by goldbattle