rpng/open_vins #141: Machine hall datasets in EuRoc

gyzou, 2021-02-18:
Is there any particular reason that machine hall datasets in EuRoc are currently not supported? I tried OpenVINS with MH_03. It can hardly track any features.

goldbattle, 2021-02-18:
I have never had that many issues with the datasets.
Normally we skip the first section of the dataset, you might want to try this script:
https://github.com/rpng/open_vins/blob/master/ov_msckf/scripts/run_ros_eth.sh#L36-L48

gyzou, 2021-02-19:
Yes, after cutting off the beginning few secs, the trajectory looks much better.

Another question: how to config for monocular mode?

Also, I am trying to make it work for KITTI datasets. So far, it does not perform as good as VINS-fusion. There are probably some parameters I don't tweak right yet. If possible, could you please take a look at KITTI dataset and see if it can be supported?

goldbattle, 2021-02-19:
Can you point me to the tools and specific datasets you are looking at?
How are you converting into a ROS bag and additionally getting the high frequency IMU information.
I would try using SLAM features, and an anchored inverse depth representation (you can take a look at the KAIST launch file).

gyzou, 2021-02-19:
The data download page is http://www.cvlibs.net/datasets/kitti/raw_data.php
The "unsyned+unrectified data" gives you 100Hz IMU data with 10 Hz image sequences.
For data format and notation, you can look at https://github.com/pratikac/kitti/blob/master/readme.raw.txt
I am mainly experimenting with 2011_09_30_drive_0018 and 2011_09_30_drive_0027 under "Residential" category
I modified some python scripts from "kitti2bag" to convert the raw data to ros bags, as they don't deal with "unsyned+unrectified data".

goldbattle, 2021-02-21:
Feel free to post your modified scripts for others that want to try with the KITTI.
Feel free to open another issue if you get KITTI working / run into issues.

closed 2021-02-21 by goldbattle

lizc4070, 2022-09-29:
@gyzou Hello, how did you solve the imu static initialization problem when using the kitti dataset？