rpng/open_vins #143: How to explain the +15 in TrackKLT?

niebayes, 2021-02-21:
What is the explanation under [this](https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_core/src/track/TrackKLT.cpp#L347) line of code? Why plus 15?

goldbattle, 2021-02-21:
The comments should explain that there was some out of bound errors that didn't seem to make sense (and increasing by a bit fixed this issue):
```c++
// Create a 2D occupancy grid for this current image
// Note that we scale this down, so that each grid point is equal to a set of pixels
// This means that we will reject points that less then grid_px_size points away then existing features
// TODO: figure out why I need to add the windowsize of the klt to handle features that are outside the image bound
// TODO: I assume this is because klt of features at the corners is not really well defined, thus if it doesn't get a match it will be out-of-bounds
Eigen::MatrixXi grid_2d = Eigen::MatrixXi::Zero((int)(img0pyr.at(0).rows/min_px_dist)+15, (int)(img0pyr.at(0).cols/min_px_dist)+15);
```

niebayes, 2021-02-22:
Another issue in the same function `TrackKLT::perform_detection_monocular`.  

At the near end of this function, two OpenCV objects are created: 
    `std::vector<cv::KeyPoint> kpts0_new; `
    `std::vector<cv::Point2f> pts0_new;`
They're used to store the tracker points, but it seems only the kpts0_new functioning. Anything I missed?

goldbattle, 2021-02-25:
Yes, they seem to be unused.

closed 2021-02-25 by goldbattle