rpng/open_vins #144: Question on Propagator::select_imu_readings

niebayes, 2021-02-22:
As `Propagator::select_imu_readings` line [201](https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/state/Propagator.cpp#L201) shows, the control block won't get into the for loop if the `imu_data.size()` is exactly `1`, and after that an empty `prop_data` will be returned.  

And in `Propagator::propagate_and_clone` line [76](https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/state/Propagator.cpp#L76), it asks for more than `2` imu measurements. Couldn't we do state propagation with less than 2 imu measurements?

WoosikLee2510, 2021-02-22:
Yes we can, but there are several reasons for not doing so:
1. IMU rate is usually 10 times faster than camera measurements, thus if it is trying to propagate 1 or 2 IMU measurement times then it is highly likely a wrong request.
2. We cut the IMU data [line](https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/state/Propagator.cpp#L238), so no matter when we want to perform the propagation we should have at least one IMU measurement. If it has less than 2 measurements then it means we did not get additional IMU measurements for propagation (up to the propagation time).
3. Actually, [line](https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/state/Propagator.cpp#L76) 76 asks for more than 1, which makes sense as said in 2.

closed 2021-02-22 by niebayes

goldbattle, 2021-02-25:
You need at least two measurements so that you can calculate a DT.
The last IMU measurement isn't used besides it timestep (refer to the discrete time equation in the propagation docs).