rpng/open_vins #145: Question on IMU pose clone ?

niebayes, 2021-02-22:
As the issue @ #135 says, the pose cloning procedure is happended at the "true" time, i.e. the current image measurement time `t_cam` compensated by the currently estimated time offset `dt`.  

However, as the line [585](https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/state/StateHelper.cpp#L585) in `StateHelper::augment_clone` shows, the cloned pose is timestamped by the image measurement time before the time offset compenstation. Would it be a problem if we do so?

WoosikLee2510, 2021-02-22:
we do the cloning at "true" time or image measurement time. For the reason. the `t_cam` compensating is done at the propagation step https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/state/Propagator.cpp#L61-L63 (by collecting IMU measurements compensating the `t_cam`), not the update step.

closed 2021-02-22 by niebayes

goldbattle, 2021-02-25:
The camera timestep is used to *map* the camera measurements from the frontend to the correct JPLPose object they should be a function of. The actual underlying state that is estimated is the true, which is defined through the augmentation step you pointed out in `StateHelper::augment_clone`. Remember that the state just has objects that point to estimates, and we can access those states using whatever access logic we want, but the Jacobians and state transition are what define the actual error states we are estimating.