rpng/open_vins #146: Question about FEJ on anchor pose?

niebayes, 2021-02-23:
As the [FEJ doc](https://docs.openvins.com/fej.html#fej-fej) shows, 

> For example if we evaluated the  H_f Jacobian with a different p_FinG at each timestep then the nullspace would not hold past the first time instance.

So the logic for the global feature parameterization is very simple:
https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/update/UpdaterHelper.cpp#L42

However, for the anchor feature parameterization, the code logic is quite complicated 

https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/update/UpdaterHelper.cpp#L87-L94

and I cannot figure out the reason why not simply use the p_FinA_fej stored in the `Feature` object. On the other hand, what does it mean by "best" feature? Is the one that generated by current cam-imu extrinsics estimate?

Can you explain the FEJ strategy further?

WoosikLee2510, 2021-02-24:
"best" feature means you use your best estimation (not fej values) to get the feature pose. The fej values may hold the nullspace logic but the value will not be as accurate as of the current estimation. In this case, the fej values of the IMU pose (R_GtoI and p_IinG ) were used.

goldbattle, 2021-02-25:
The comment should provide some insight:
```
// Calculate the position of this feature in the global frame FEJ
// If anchored, then we can use the "best" p_FinG since the value of p_FinA does not matter
```
As Woosik said, the best just refers to the most recent estimate of the feature, and thus is the current best estimate. For anchored we found that the choice of the pFinA does not seem to affect consistency (you can derive the nullspace to see for yourself), thus we choose it such that the global pFinG is the current best estimate. This is complicated a bit since we are using the Jacobian chainrule, thus we need to ensure that both the local anchor estimate Jacobian which is the derivative of pFinG in respect to pFinA uses the same pFinA/G estimate as the top level chainrule Jacobian which is the derivative of pFinC in respect to pFinG.

Additionally, I recommend you use the simulator to test out what you suggested yourself to see how it performs.

niebayes, 2021-02-26:
Thanks, I'll try it!

closed 2021-02-26 by niebayes