rpng/open_vins #147: L515 camera drifts away really fast

RapIt, 2021-02-26:
Hi,
thank you for publishing your great work. I know there are many posts on the correct calibration procedure but I am struggeling to get it work with a Realsense L515 camera (only using its monocular RGB + IMU sensor for now). I already spend some time adjusting all the parameters but in so far I always found the system to diverge right after IMU excitement. Odometry says something about 100m/second:

[image]

I used both the calibration from the realsense package as well as manual calibration results. I also tried different camera resolutions. For the `C0_to_I` transform I used the extrinsics provided by the realsense driver using `rs-enumerate-devices -c`

I think it might be more of a camera than an IMU issue.
Right after startup the feature extractor seems to work:
```
[TIME]: 0.0005 seconds for MSCKF update (4 feats)
[TIME]: 0.0002 seconds for SLAM update (9 feats)
[TIME]: 0.0015 seconds for SLAM delayed init (3 feats)
```
I also verified with the `test_tracking` script. After some small time interval I don't get any features anymore even if the scene barely changed:
 ```
[TIME]: 0.0001 seconds for MSCKF update (0 feats)
[TIME]: 0.0000 seconds for SLAM update (0 feats)
[TIME]: 0.0003 seconds for SLAM delayed init (0 feats)
[TIME]: 0.0010 seconds for re-tri & marg (11 clones in state)
```

Relevant parts of the launch file are:

        <param name="topic_imu"      type="string" value="/camera/imu" />
        <param name="topic_camera0"  type="string" value="/camera/color/image_raw" />

        
        <param name="use_fej"                type="bool"   value="true" />
        <param name="use_imuavg"             type="bool"   value="true" />
        <param name="use_rk4int"             type="bool"   value="true" />
        <param name="use_stereo"             type="bool"   value="$(arg use_stereo)" />
        <param name="calib_cam_extrinsics"   type="bool"   value="true" />
        <param name="calib_cam_intrinsics"   type="bool"   value="true" />
        <param name="calib_cam_timeoffset"   type="bool"   value="true" />
        <param name="calib_camimu_dt"        type="double" value="0.0" />
        <param name="max_clones"             type="int"    value="11" />
        <param name="max_slam"               type="int"    value="50" />
        <param name="max_slam_in_update"     type="int"    value="25" /> 
        <param name="max_msckf_in_update"    type="int"    value="999" />
        <param name="max_cameras"            type="int"    value="$(arg max_cameras)" />
        <param name="dt_slam_delay"          type="double" value="5" />

        
        <param name="use_klt"            type="bool"   value="true" />
        <param name="num_pts"            type="int"    value="200" />
        <param name="fast_threshold"     type="int"    value="15" />
        <param name="grid_x"             type="int"    value="5" />
        <param name="grid_y"             type="int"    value="3" />
        <param name="min_px_dist"        type="int"    value="5" />
        <param name="knn_ratio"          type="double" value="0.65" />
        <param name="downsample_cameras" type="bool"   value="false" />

        <param name="up_msckf_sigma_px"            type="double"   value="1" />
        <param name="up_msckf_chi2_multipler"      type="double"   value="1" />
        <param name="up_slam_sigma_px"             type="double"   value="1" />
        <param name="up_slam_chi2_multipler"       type="double"   value="1" />
        <param name="up_aruco_sigma_px"            type="double"   value="1" />
        <param name="up_aruco_chi2_multipler"      type="double"   value="1" />
        <param name="gyroscope_noise_density"      type="double"   value="1.6968e-04" />
        <param name="gyroscope_random_walk"        type="double"   value="1.9393e-05" />
        <param name="accelerometer_noise_density"  type="double"   value="2.0000e-3" />
        <param name="accelerometer_random_walk"    type="double"   value="3.0000e-3" />

        
        <rosparam param="cam0_wh">[1280, 720]</rosparam>
        <param name="cam0_is_fisheye" type="bool" value="false" />
        
        
        <rosparam param="cam0_k">[912.345845, 912.883284, 627.872056, 384.857433]</rosparam>
        <rosparam param="cam0_d">[0.106914, -0.191032, 0.005422, -0.001663, 0.000000]</rosparam>
        
        <rosparam param="T_C0toI">
            [
            0.999918, -0.0123185, 0.00337064, -0.0113789290189743,
            0.0123902, 0.999678, -0.0221653, 0.00240374635905027,
            -0.00309651, 0.0222053, 0.999749, 0.00564641039818525,
            0.0, 0.0, 0.0, 1.0
            ]
        </rosparam>

I didn't spend too much time on the IMU noise tuning as I think it is more of a camera problem, so I just grabbed the defaults from eth launch file. 

Any hint on what might look suspicious would be greatly appreciated! If someone else made it work with the L515 I would love to see a launchfile.

goldbattle, 2021-02-26:
Your distortion parameters should only be 4 in length.
Make sure you look at the console printout to make sure everything is being read in correctly.
I also suggest you try calibrating with Kalibr, to see if you get similar values that you have here.
If it isn't getting tracks, then it might be the camera intrinsics since those are used for the RANSAC process in KLT tracking.

Most failures are due to people putting the wrong transformation in (we require rotation from C0 to I, and position of I in C0).

RapIt, 2021-02-26:
> Your distortion parameters should only be 4 in length.

Yes indeed, I pasted them from the ros `camera_calibration` package. But it makes no difference, the parameters are read correctly on startup. I again performed an intrinsics calibration with kalibr but the parameters are quite similar:
```
<rosparam param="cam0_k">[921.01305021, 921.14028947, 628.71924433, 398.34314184]</rosparam>
<rosparam param="cam0_d">[0.08279712, -0.21664636,  0.00424575, -0.00353133]</rosparam>
<rosparam param="T_C0toI">
            [
            0.999918, -0.0123185, 0.00337064, 0.0113789290189743,
            0.0123902, 0.999678, -0.0221653, -0.00240374635905027,
            -0.00309651, 0.0222053, 0.999749, -0.00564641039818525,
            0.0, 0.0, 0.0, 1.0
            ]
</rosparam>

```
> Most failures are due to people putting the wrong transformation

Then I guess I am one of them. I inverted the translational part as this was also C0 to I:
```
rs-enumerate-devices -c 

Extrinsic from "Color"	  To	  "Accel" :
 Rotation Matrix:
   0.999918        -0.0123185        0.00337064    
   0.0123902        0.999678        -0.0221653     
  -0.00309651       0.0222053        0.999749      

 Translation Vector: -0.0113789290189743  0.00240374635905027  0.00564641039818525  
```

> Make sure you look at the console printout to make sure everything is being read in correctly.

As far as I can tell the parameters on startup look the same to the ones I provided:
```
cam_0_fisheye:0
cam_0_wh:
1280 x 720
cam_0_intrinsic(0:3):
921.013  921.14 628.719 398.343
cam_0_intrinsic(4:7):
  0.0827971   -0.216646  0.00424575 -0.00353133
cam_0_extrinsic(0:3):
 0.0110936 0.00161692 0.00617768   0.999918
cam_0_extrinsic(4:6):
-0.0113657 0.00266852 0.00555336
T_C0toI:
   0.999918  -0.0123185  0.00337064   0.0113789
  0.0123902    0.999678  -0.0221653 -0.00240375
-0.00309651   0.0222053    0.999749 -0.00564641
          0           0           0           1
```
So far I can see no difference, the trajectory is still diverging in a couple of seconds.
From what I can see in rviz, there are quite a lot "active features" but the number of SLAM and MSCKF features is very low and after some time always zero.

goldbattle, 2021-03-08:
If that is the case, then it might be the chi2 rejecting features, so it might be due to the IMU noises / IMU measurement quality. I would still suggest you try to calibrate with Kalibr if you can, as this can raise any issues with the camera / imu before you even try with OpenVINS. Additionally, I am not sure if this is like the T265, but if the gyroscope and accelerometer publish at different rates, you need to enable "fusion" of them in the realsense driver (which one are you using?).

RapIt, 2021-03-11:
Thank you @goldbattle, the intrinsic camera calibration seemed good with kalibr as well. But as you stated correctly, IMU measurement quality was miserable. Apparently accel part of the IMU was sending crazy high values leading to high noise values when plotting allan variance. This is only when using the `linear_interpolation` method in the realsense-ros driver to fuse values to a single imu topic and probably due to a timestamping bug in the realsense-driver. I didn't notice as I enabled this option only in this specific launchfile.

Thank you for your help, I will close this for now and try open-vins again when I got a working camera/imu setup.

closed 2021-03-11 by RapIt

goldbattle, 2021-03-12:
I have used the `linear_interpolation` with the T265 sensor, and it seemed to work ok (but not the best compared to an external IMU). This seems to be the same issue that the D435i has (see #102).