rpng/open_vins #149: Camera processing questions

mhuzai, 2021-03-03:
Hi, I had a couple of questions regarding how camera frames are handled:

1. Why is [this mutex](https://github.com/rpng/open_vins/blob/16f5c37f2f9a13f6641f160e2355a763315327ba/ov_core/src/track/TrackBase.h#L364) required when processing camera images? Doesn't open_vins process one camera frame at a time? Or is this to support the case when processing runs over the deadline and a new camera frame arrives in parallel?
2. I had always been confused by how camera images were handled in `ros_serial_msckf.cpp`. Upon the arrival of a new camera frame for time `t`, frame `t-1` would be fed into the system. I noticed that this was fixed in [this commit](https://github.com/rpng/open_vins/commit/0c6a8f7946dda33db4f75712db788d4b9e97c211#diff-9438b9d30362e5b7f50f63693ecc50117f78f528324ba70c961efff344650df0), but I was wondering if you could explain the original implementation, and what enabled the new one with "current" images being fed into the system.

Thank you!

WoosikLee2510, 2021-03-04:
For the second question, we kept one camera image in the buffer to ensure we have enough IMU measurements to propagate the state (it was kept in `ros_serial_msckf.cpp`). Now the buffer is handled by `VioManager.h` and the camera image will be processed whenever IMU can propage.
https://github.com/rpng/open_vins/blob/0c6a8f7946dda33db4f75712db788d4b9e97c211/ov_msckf/src/core/VioManager.cpp#L177-L181

goldbattle, 2021-03-08:
Yes, that mutex it to allow for the case that you want to process a bunch of monocular feeds in parallel. The binocular is an example of this were we process in parallel and thus the trackers, and feature database need to handle this parallel case.

As Woosik states, we use to buffer one image so that we ensure that have enough IMU measurements to propagate to the imaging time. This is due to the estimation of the camera-imu time offset which means that we might not have all IMU measurements that we need for propagation when we get the image. We now recently switched over to the queue, so this is hidden from the user, but we still do this buffering for simulation for example.

mhuzai, 2021-03-08:
Great, thank you!

closed 2021-03-08 by mhuzai