rpng/open_vins #150: Left and Right Features in Stereo

norabarlow, 2021-03-10:
Hello!

Why is Lucas-Kanade optical flow used to match features in the left and right images instead of something simpler, such as epipolar block matching? Does it lead to more accurate results?

https://github.com/rpng/open_vins/blob/master/ov_core/src/track/TrackKLT.cpp#L488

Thanks!

goldbattle, 2021-03-10:
This is for reduces computation by preventing the need for extraction of descriptors which would cost more. Ideally, we would do descriptor matching from left to right images, but here the assumption is that features have large depths (in respect to the baseline of the stereo camera), thus they should fall in a similar location in the right image. Basically we are treating the left to right KLT as the epipolar searching here. Some of the comments should provide insight:
```
// This does the same logic as the perform_detection_monocular() function, but we also enforce stereo constraints.
// So we detect features in the left image, and then KLT track them onto the right image.
// TODO: Project points from the left frame into the right frame
// TODO: This will not work for large baseline systems.....
// Do our KLT tracking from the left to the right frame of reference
// Note: we have a pretty big window size here since our projection might be bad
// Note: but this might cause failure in cases of repeated textures (eg. checkerboard)
```

Since we track on the undistorted images (to allow calibration of the camera intrinsics and distortion parameters), an epipolar search would have to be the general one, and is not as simple as traditional block matching but could be doable. Hope this helps.

norabarlow, 2021-03-11:
Makes sense, thank you for the explanation!

closed 2021-03-12 by goldbattle