rpng/open_vins #152: T_C0to_I and T_C1to_I for TUM VI Dataset

norabarlow, 2021-03-18:
Hello!

I want to run OpenVINS on the 1024x1024 version of the TUM VI dataset. I noticed you already had a launch file for 512x512 images, which has been very helpful in creating a launch file for the larger image sizes.

I am trying to update the camera calibration parameters for 1024x1024 but couldn't find where you got the values for the last column of values in `T_C0to_I` and `T_C1to_I` for the 512x512 images. The other values I found in `camchain.yaml` in the TUM VI dataset. Would you be able to tell me how you got those last column of values? It could very well be something simple that I am missing.

Thanks!

goldbattle, 2021-03-22:
The `results-imucam-imucalib.txt` from Kalibr is what we always pull the translation from, these are provided in their zip downloads from their website:
- https://vision.in.tum.de/_media/data/datasets/visual-inertial-dataset/pinhole-equi-1024.zip
- https://vision.in.tum.de/_media/data/datasets/visual-inertial-dataset/pinhole-equi-512.zip

The `T_ic:  (cam0 to imu0)` 4x4 transform is what is in the [launch file](https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_ros_tum.launch) and contains the rotation from the cam0 frame to the imu, and the position of the cam0 frame seen/in the imu frame. If you don't have this value, you can also inverse the one provided in the camchain.yaml (i.e. T_C0toI = T_ci^-1 in Kalibr's files).

NikolausDemmel, 2021-03-22:
Note that camera and IMU extrinsics should be practically the same for both the 1024 and 512 version in TUMVI. It's simply downsampled for convenience, so all you need to adapt are the camera intrinsic parameters.

norabarlow, 2021-03-26:
Thanks for the info!

closed 2021-03-29 by goldbattle