rpng/open_vins #153: Pose covariance output

johanndiep, 2021-03-18:
Does the current ROS framework output EKF covariances as well?

If not, can someone point me to the source code where those covariances are calculated (and maybe saved)?

Thank you very much in advance!

WoosikLee2510, 2021-03-18:
Where cov is published:
https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/core/RosVisualizer.cpp#L200-L208
where cov is saved:
https://github.com/rpng/open_vins/blob/e516f7c564c2159b7edfb42a8230b732bf6aebc2/ov_msckf/src/state/State.h#L124

johanndiep, 2021-03-19:
Thank you for the hint. Note on the comment:

// TODO: this currently is an approximation since this should actually evolve over our propagation period
// TODO: but to save time we only propagate the mean and not the uncertainty, but maybe we should try to prop the covariance?

Is the covariance actually calculated according to the EKF?

yangyulin, 2021-03-19:
Yes.
The EKF will keep the updated state and cov.
When plotting or doing a Mahalanobis test, we just grab the related cov
from the state cov matrix.

closed 2021-03-19 by johanndiep