rpng/open_vins #154: Extrinsic Parameters coordinate frames

rakshith95, 2021-03-23:
Hello, 
In the documentation for the measurement update (https://docs.openvins.com/update-feat.html#relative), 
[image]

Why is the extrinsic parameters (Rotation, translation) defined this way? Afaik the standard is to define them as transforms from a World frame ---> Camera frame, which would mean that Rotation and translation are defined as the pose of the global frame in the camera coordinate frame.

goldbattle, 2021-03-25:
When you say "extrinsics" you are referring to the current frame estimate and not the sensor extrinsics I assume.
This is due to the convention of the original MSCKF paper's state definition and all works which derive from the original paper.
It doesn't make much a difference as long as you are consistent in how you take your derivatives and update those state elements.

rakshith95, 2021-03-25:
@goldbattle 
I was referring to the line in the picture above which says "{R,p} denote the current camera pose and orientation in the global frame (or camera extrinsics)" 
I thought this meant that the sensor extrinsics were defined as such.

goldbattle, 2021-03-26:
The pose is a function of the camera extrinsics (underlined in red):
[image]

rakshith95, 2021-03-28:
Alright, thank you. Closing this

closed 2021-03-28 by rakshith95