rpng/open_vins #156: Reset estimator without restarting OpenVins

TheRealKaufmann, 2021-03-29:
Hi there,

I am currently testing OpenVins on a UAV System for visual odometry. It works really well outdoors. I am fusing th output of open vins into another EKF-estimator that uses GPS as one input source.
During high altitude flights (>20m) sometimes the OpenVins estimator gets lost and drifts away. I already implemented a feature to detect this behavior. 

My Question: Is it possible to simply „reset“ the open vins estimator if I detect that it gets lost? 

Then my estimator could rely on gps for a while until the open vins output seems okay again. (Since the OpenVins output is fused into another system, it is not really important that the OpenVins output is continuous.)

Also: If you know of a better way to deal with this type of problem then please let me know. 

Thanks!
Tom

WoosikLee2510, 2021-03-29:
Currently, OpenVINS is not supporting the `reset` function. One of the reasons is that OpenVINS initialization requires standing still motion which is a very effective motion assumption for initializing IMU bias. 
So, in order to (re)initialize this vio system during flight, you need to have different initialization techniques such as structure-from-motion. This must be in our TODO list but not finished yet.

closed 2021-04-02 by TheRealKaufmann

mzahana, 2024-05-27:
I am also looking for this re-initialization feature. Is it implemented in OpenVINS or not yet?
@TheRealKaufmann  Have you figured out a way to re-initialize the estimator ?