rpng/open_vins #157: open_vins on runoff issue on zed2

truncs, 2021-04-09:
Hi,

I am trying to run open_vins using zed2 cameras and testing it on a bag results in very unstable results (lot of bias in the trajectory and can't detect stop velocity). I am attaching the screenshot and the config files that I use 

[image]

```
<launch>

    <node name="run_serial_msckf" pkg="ov_msckf" type="run_serial_msckf" output="screen">
    
        <!-- bag topics -->
        <param name="topic_imu"      type="string" value="/zed2/zed_node/imu/data_raw" />
        <param name="topic_camera0"  type="string" value="/zed2/zed_node/left_raw/image_raw_color" />
        <param name="topic_camera1"  type="string" value="/zed2/zed_node/right_raw/image_raw_color" />
        <rosparam param="stereo_pairs">[0,1]</rosparam>
    
        <!-- bag parameters -->
        <param name="path_bag"    type="string" value="/root/data/zed_20210319104926.bag" />
        <param name="bag_start"   type="int"    value="0" />
        <param name="bag_durr"    type="int"    value="-1" />
    

	<!-- world/filter parameters -->
        <param name="use_fej"                type="bool"   value="true" />
        <param name="use_imuavg"             type="bool"   value="true" />
        <param name="use_rk4int"             type="bool"   value="true" />
        <param name="use_stereo"             type="bool"   value="true" />
        <param name="calib_cam_extrinsics"   type="bool"   value="true" />
        <param name="calib_cam_intrinsics"   type="bool"   value="true" />
        <param name="calib_cam_timeoffset"   type="bool"   value="true" />
        <param name="calib_camimu_dt"        type="double" value="0.017971296601266912" />
        <param name="max_clones"             type="int"    value="11" />
        <param name="max_slam"               type="int"    value="75" />
        <param name="max_slam_in_update"     type="int"    value="25" /> <!-- 25 seems to work well -->
        <param name="max_msckf_in_update"    type="int"    value="40" />
        <param name="max_cameras"            type="int"    value="2" />
        <param name="dt_slam_delay"          type="double" value="3" />

        <param name="init_window_time"       type="double" value="0.5" />
        <param name="init_imu_thresh"        type="double" value="0.1" />
        <rosparam param="gravity">[0.0,0.0,9.81]</rosparam>
        <param name="feat_rep_msckf"         type="string" value="GLOBAL_3D" />
        <param name="feat_rep_slam"          type="string" value="GLOBAL_3D" />

	<!-- zero velocity update parameters -->
        <param name="try_zupt"               type="bool"   value="false" />
        <param name="zupt_chi2_multipler"    type="int"    value="2" />
/        <param name="zupt_max_velocity"      type="double" value="0.3" />
        <param name="zupt_noise_multiplier"  type="double" value="50" />
		 
        <!-- tracker/extractor parameters -->
        <param name="use_klt"          type="bool"   value="true" />
        <param name="fast_threshold"   type="int"    value="10" />
        <param name="grid_x"           type="int"    value="5" />
        <param name="grid_y"           type="int"    value="3" />
        <param name="min_px_dist"      type="int"    value="10" />
        <param name="num_pts"          type="int"    value="400" />
    
        <!-- sensor noise values / update -->
        <param name="up_msckf_sigma_px"            type="double"   value="5" />
        <param name="up_msckf_chi2_multipler"      type="double"   value="1" />
        <param name="gyroscope_noise_density"      type="double"   value="0.0002" />
        <param name="gyroscope_random_walk"        type="double"   value="" />
        <param name="accelerometer_noise_density"  type="double"   value="0.02" />
        <param name="accelerometer_random_walk"    type="double"   value="8.0e-05" />
    
        <!-- camera intrinsics -->
	<rosparam param="cam0_wh">[1280, 720]</rosparam>
        <rosparam param="cam1_wh">[1280, 720]</rosparam>
[... 116 lines cut ...]
      <param name="noise/feature" value="0.035"/>

      <param name="initial_state/velocity/x" value="0.0"/>
      <param name="initial_state/velocity/y" value="0.0"/>
      <param name="initial_state/velocity/z" value="0.0"/>

      <!-- These values should be covariance -->
      <param name="initial_covariance/velocity" value="0.25"/>
      <param name="initial_covariance/gyro_bias" value="0.01"/>
      <param name="initial_covariance/acc_bias" value="0.01"/>
      <param name="initial_covariance/extrinsic_rotation_cov" value="3.0462e-4"/>
      <param name="initial_covariance/extrinsic_translation_cov" value="2.5e-5"/>

      <remap from="~imu" to="/zed2/zed_node/imu/data_raw"/>
      <remap from="~features" to="image_processor/features"/>

    </node>
  </group>

</launch>
```

I am wondering if there are any thoughts about what could be going wrong or how to go about debugging this?

Thanks!

goldbattle, 2021-04-12:
The most common issue people run into is that we require the camera to IMU transformation. Pull this value from the result.txt file that Kalibr generates. Additionally the xml you posted has some syntax errors you need to fix. For your IMU noise you should start with the default ones specified in the eurocmav launch file before trying your own. If you are using your own, you shouldn't set them to zero as you have. Your instrinsics says that the two cameras are separated by 5cm, is this reasonable?

You will likely need to tune the `init_window_time` and `init_imu_thresh`  please see this issue along with the other closed issues: https://github.com/rpng/open_vins/issues/93#issuecomment-678667821

truncs, 2021-04-13:
Ah! Completely missed the part on camera to IMU transformation. Now it is working fairly well (screenshot attached).

[image]

Thanks for the help!

closed 2021-04-13 by truncs