rpng/open_vins #158: [Question]: What are the supported camera projection models?

rakshith95, 2021-04-11:
Hello,

Kalibr supports the following models as given [here](https://github.com/ethz-asl/kalibr/wiki/supported-models):
pinhole camera model (pinhole)
omnidirectional camera model (omni)
double sphere camera model (ds)
extended unified camera model (eucm)

But I couldn't find any such list of supported camera models in the documentation of Open VINS. 
Only the option of fisheye, or radtan distortion models are present; does that mean that the only supported model is the pinhole + distortion model?

goldbattle, 2021-04-12:
We support the `pinhole` camera model and the `radial-tangential` and `equidistant` fisheye models for the distortion parameters. The exact details of the distortion models are described here:
https://docs.openvins.com/update-feat.html#distortion

rakshith95, 2021-04-12:
Alright, thank you

closed 2021-04-12 by rakshith95