rpng/open_vins #160: ov_msckf run off at start. Possible IMU issue

rakshith95, 2021-04-13:
Hello,
I'm trying to run open vins with topics input from a gazebo simulation envt, using the `ros_subscribe_msckf`. 
My issue is similar to #157, in that the odometry seems to completely go bonkers soon after the beginning. In that, it's mentioned that 
"You will likely need to tune the init_window_time and init_imu_thresh"
I have tuned my init_imu_thresh parameter, such that it initializes right as the drone, takes off the filter is initialized. However, I am not sure what the init_window_time parameter actually does and how it must be tuned.

`
<launch>

    
    <arg name="max_cameras" default="1" />
    <arg name="use_stereo"  default="false" />

    
    <arg name="init_window_time"  default="0.5" />
    <arg name="init_imu_thresh"   default="0.075" />

    
    
    
    
    /

    
    <node name="run_subscribe_msckf" pkg="ov_msckf" type="run_subscribe_msckf" output="screen" clear_params="true" required="true">

        <param name="topic_imu"      type="string" value="/uav1/vio/imu" />
        <param name="topic_camera0"  type="string" value="/uav1/vio/camera/image_raw/" />
        
        
        <param name="use_aruco"              type="bool"   value="false" />
        <param name="use_fej"                type="bool"   value="true" />
        <param name="use_imuavg"             type="bool"   value="true" />
        <param name="use_rk4int"             type="bool"   value="true" />
        <param name="use_stereo"             type="bool"   value="$(arg use_stereo)" />
        <param name="calib_cam_extrinsics"   type="bool"   value="true" />
        <param name="calib_cam_intrinsics"   type="bool"   value="true" />
        <param name="calib_cam_timeoffset"   type="bool"   value="true" />
        <param name="calib_camimu_dt"        type="double" value="0" />
        <param name="max_clones"             type="int"    value="15" />
        <param name="max_slam"               type="int"    value="75" />
        <param name="max_slam_in_update"     type="int"    value="25" />
        <param name="max_msckf_in_update"    type="int"    value="40" />
        <param name="max_cameras"            type="int"    value="$(arg max_cameras)" />
        <param name="dt_slam_delay"          type="double" value="3" />
        <param name="init_window_time"       type="double" value="$(arg init_window_time)" />
        <param name="init_imu_thresh"        type="double" value="$(arg init_imu_thresh)" />
        <rosparam param="gravity">[0.0,0,9.81]</rosparam>
        <param name="feat_rep_msckf"         type="string" value="GLOBAL_3D" />
        <param name="feat_rep_slam"          type="string" value="ANCHORED_INVERSE_DEPTH_SINGLE" /> 

        
        <param name="try_zupt"               type="bool"   value="false" />
        <param name="zupt_chi2_multipler"    type="int"    value="2" />
        <param name="zupt_max_velocity"      type="double" value="0.5" />
        <param name="zupt_noise_multiplier"  type="double" value="50" />

        

        
        
        
        <param name="use_klt"            type="bool"   value="true" />
        <param name="num_pts"            type="int"    value="500" />
        <param name="fast_threshold"     type="int"    value="15" />
        <param name="grid_x"             type="int"    value="5" />
        <param name="grid_y"             type="int"    value="3" />
        <param name="min_px_dist"        type="int"    value="5" />
        <param name="knn_ratio"          type="double" value="0.70" />
        <param name="downsample_cameras" type="bool"   value="false" />
        <param name="multi_threading"    type="bool"   value="true" />

        
        <param name="up_msckf_sigma_px"            type="double"   value="1" />
        <param name="up_msckf_chi2_multipler"      type="double"   value="1" />
        <param name="up_slam_sigma_px"             type="double"   value="1" />
        <param name="up_slam_chi2_multipler"       type="double"   value="1" />

        <param name="gyroscope_noise_density"      type="double"   value="0.3" />
        <param name="gyroscope_random_walk"        type="double"   value="0.001" />
        <param name="accelerometer_noise_density"  type="double"   value="0.1" />
        <param name="accelerometer_random_walk"    type="double"   value="0.001" />

        
        <rosparam param="cam0_wh">[752, 480]</rosparam>
        <param name="cam0_is_fisheye" type="bool" value="true" />

        <rosparam param="cam0_k">[227.4010064226358,227.35879407313246,375.5302935901654,239.4881944649193]</rosparam>
        <rosparam param="cam0_d"> [0.019265981371039506, 0.0011428473998276235, -0.0003811659324868097, 6.340084698783884e-05]</rosparam>
        <rosparam param="T_C0toI">
            [
            1, 0, 0, 0,
            0, 1, 0, 0,
            0, 0, 1, 0,
            0, 0, 0, 1
            ]
        </rosparam>

    </node>

    
    

    
    <node type="rviz" name="rviz" pkg="rviz" args="-d $(find ov_msckf)/launch/display.rviz" />

</launch>
`

I have posted the launch file I am using. Could you please let me know if I should make any other changes?

Thank you

goldbattle, 2021-04-13:
Are you able to pose a bag file (via google drive or other service)?
The window is used to calculate the direction of gravity, and 0.75-1 second
is what I normally use.
If you change this then you will need to re-tune, but you shouldn't need to
tune per-dataset in general.

I also suggest you try using the default ETH imu noise values before trying
your own.
Additionally, calibrate with Kalibr to ensure success (I see that your
camera imu extrinsic is just the identity which can never be the case
outside of simulation).

rakshith95, 2021-04-14:
Hello,
Thank you for the suggestions @goldbattle .  [Here](https://drive.google.com/file/d/1DY4fBnY0VuF95W1bpnPs8W6Rb8fOHxLv/view?usp=sharing) is the link for a bagfile I recorded now where the drone takes off from a starting position, and moves a bit.  

I tried values of the  `init_window` from 0.75-1 but none worked well, which is why I thought I'd try different values. 
I did try the default ETH IMU noise values, but the same was happening. I am basing these values, from a config file I was using for VINS-Mono which worked well. I will post that as well for your reference:

```
%YAML:1.0

#common parameters
imu_topic: "/uav1/vio/imu"
image_topic: "/uav1/vio/camera/image_raw"
output_path: "/home/honzabednar/git/vins-mono/output/"

model_type: KANNALA_BRANDT
camera_name: camera
image_width: 752
image_height: 480
projection_parameters:
   k2: 0.019265981371039506
   k3: 0.0011428473998276235
   k4: -0.0003811659324868097
   k5: 6.340084698783884e-05 
   mu: 227.4010064226358
   mv: 227.35879407313246
   u0: 375.5302935901654
   v0: 239.4881944649193

# Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 0   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.
                        # 2  Don't know anything about extrinsic parameters. You don't need to give R,T. We will try to calibrate it. Do some rotation movement at beginning.                        
#If you choose 0 or 1, you should write down the following matrix.
#Rotation from camera frame to imu frame, imu^R_cam

# REALSENSE FRONT
# rosrun tf tf_echo uav1/fcu uav1/rs_d435/infra1_optical
#extrinsicRotation: !!opencv-matrix
#   rows: 3
#   cols: 3
#   dt: d
#   data: [0.0, 0.0, 1.0,
#         -1.0, 0.0, 0.0,
#          0.0,-1.0, 0.0]
##Translation from camera frame to imu frame, imu^T_cam
#extrinsicTranslation: !!opencv-matrix
#   rows: 3
#   cols: 1
#   dt: d
#   data: [0.155, 0.018,-0.089]
   

# REALSENSE DOWN
#extrinsicRotation: !!opencv-matrix
#   rows: 3
#   cols: 3
#   dt: d
#   data: [0.0,-1.0, 0.0,
#         -1.0, 0.0, 0.0,
#          0.0, 0.0,-1.0]
##Translation from camera frame to imu frame, imu^T_cam
#extrinsicTranslation: !!opencv-matrix
#   rows: 3
#   cols: 1
#   dt: d
#   data: [0.05, 0.0, -0.093]

# BLUEFOX FISHEYE
extrinsicRotation: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [1.0, 0.0, 0.0,
          0.0, 1.0, 0.0,
          0.0, 0.0, 1.0]
#Translation from camera frame to imu frame, imu^T_cam
extrinsicTranslation: !!opencv-matrix
   rows: 3
   cols: 1
   dt: d
   data: [0.0, 0.0, 0.0]

#Multiple thread support
multiple_thread: 1

#feature traker paprameters
max_cnt: 100            # max feature number in feature tracking
min_dist: 30            # min distance between two features 
freq: 10                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image 
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 1           # publish tracking image as topic
flow_back: 0            # perform forward and backward optical flow to improve feature tracking accuracy
equalize: 0             # if image is too dark or light, trun on equalize to find enough features
fisheye: 1              # if using fisheye, turn on it. A fisheye_mask_name has to be specified and placed into /config folder
fisheye_mask_name: "fisheye_mask_752x480.jpg"              # name of the mask file 

#optimization parameters
max_solver_time: 0.04  # max solver itration time (ms), to guarantee real time
max_num_iterations: 10   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.1          # accelerometer measurement noise standard deviation. #0.2   0.04
gyr_n: 0.3         # gyroscope measurement noise standard deviation.     #0.05  0.004
acc_w: 0.001         # accelerometer bias random work noise standard deviation.  #0.02
gyr_w: 0.001       # gyroscope bias random work noise standard deviation.     #4.0e-5
g_norm: 9.81007     # gravity magnitude

#loop closure parameters
loop_closure: 0                    # start loop closure
load_previous_pose_graph: 0        # load and reuse previous pose graph; load from 'pose_graph_save_path'
fast_relocalization: 1            # useful in real-time and large project
pose_graph_save_path: "/home/honzabednar/git/vins-mono/output/pose_graph/" # save and load path

#unsynchronization parameters
estimate_td: 1                      # online estimate time offset between camera and imu
td: 0.0                             # initial value of time offset. unit: s. readed image clock + td = real image clock (IMU clock)

#rolling shutter parameters
rolling_shutter: 0                  # 0: global shutter camera, 1: rolling shutter camera
rolling_shutter_tr: 0               # unit: s. rolling shutter read out time per frame (from data sheet). 

#visualization parameters
save_image: 0                   # save image in pose graph for visualization prupose; you can close this function by setting 0 
visualize_imu_forward: 1        # output imu forward propogation to achieve low latency and high frequence results
visualize_camera_size: 0.4      # size of camera marker in RVIZ

```

Since I am testing it completely in the simulation environment for now, I thought the extrinsic calibration of Identity, should be fine.

rakshith95, 2021-04-15:
Hi @goldbattle. Do you have any suggestions after seeing the bag file?

goldbattle, 2021-04-15:
Sorry I have not had time, hopefully I can take a look this weekend / next
week when I find some time.
Thanks again for posting it.

rakshith95, 2021-04-15:
Sure, thank you!

rakshith95, 2021-04-19:
Hello @goldbattle, sorry to bother you, but did you have a chance to check this?

goldbattle, 2021-06-16:
It looks like the main issue is that the platform stops part way through the dataset which is a common problem for all VIO systems. You can experiment with this launch file:
https://gist.github.com/goldbattle/165055dab7e2c79c87a9e4392f0c3744

rakshith95, 2021-06-20:
Hello @goldbattle , thanks for getting back on this. I will try it with the launch file provided.
However, I'm not sure what you mean by " the main issue is that the platform stops part way through the dataset ". Could you expand more on that?

goldbattle, 2021-06-21:
You should experiment with enabling the zero velocity update to handle standstill. When the platform is stationary this is a degenerate motion for any VO / VIO platform. For example if you are stationary, are you able to triangulate 3d features in the environment?

rakshith95, 2021-06-22:
Okay, that makes sense thank you.  I'm closing the issue for now since with the launch file you've provided it works without the run off,  if I disable the zero velocity update.

closed 2021-06-22 by rakshith95