rpng/open_vins #162: bad positive semi-definitate of Cov

gao-ouyang, 2021-04-21:
I try openvins using my dataset,  i find that positive semi-definitate of Cov is bad once in a while.   
it only happen when perform_anchor_change() and EKFPropagation(). we cannot find why it is. do you have any idea?

_LOG: StateHelper::EKFPropagation() - diagonal at 34 is -1.845405e-08 in EKFPropagation()_

goldbattle, 2021-04-21:
This can be caused by a lot of things, but typically it is due to poor noise selection or outliers which pass the chi2 check.

gao-ouyang, 2021-04-22:
Thank you for your generous instruction.
I checked the question again, i find that it is caused by EKFUpdate(),  I calculate Cov's eigenvalue in this Update, the smallest eigenvalue is about -1e-9, it's bad(In general, it is about -1e-18).
i have two question:
1) if not handle positive semi-definitate, whether it will lower EKF's accurracy
2) to keep positive semi-defintate, do you have any idea?

goldbattle, 2021-05-29:
You cannot perform filtering without a positive semi-definite (PSD) matrix from the definition of the covariance matrix (see [this](https://home.ttic.edu/~dmcallester/ttic101-07/lectures/Gaussians/Gaussians.pdf) for a refresher). You can become non-PSD if there are numerical issues or the noises selected are too small (thus the covariance goes towards zero and will run into numerical issues). I recommend you play with your noises, and if you are making a custom update function then it is likely your update causing it. It is just the current master branch, then it is likely poor chose of sensor noises or the system diverging from poor initialization. Hope this helps.

goldbattle, 2021-06-09:
Feel free to re-open if needed.

closed 2021-06-09 by goldbattle