rpng/open_vins #163: Run EuRoC with images instead of ros bag

CanCanZeng, 2021-04-23:
Hi, I try to run open_vins without ros bag, but its performance is a slightly worse then the result from rogbag, I do not know how this come. The configurations are the same. I suspect the only difference is in feeding data to _VioManager_ , and my way is like below
```

for(size_t idx_frame = idx_start; idx_frame < idx_end; idx_frame++) {
        double t_img = img_names_vec[idx_frame].first;

        std::vector<ov_core::ImuData> measure_imu_vec;
        while (iter_imu_data != imu_data_vec.end() && iter_imu_data->first < t_img) {
            auto imu_data = iter_imu_data->second;
            ov_core::ImuData measure_imu;
            measure_imu.timestamp = iter_imu_data->first;
            measure_imu.am << imu_data[3], imu_data[4], imu_data[5];
            measure_imu.wm << imu_data[0], imu_data[1], imu_data[2];

            measure_imu_vec.push_back(measure_imu);
            iter_imu_data++;
        }

        for(const auto& measure_imu : measure_imu_vec) {
            sys->feed_measurement_imu(measure_imu);
        }

        cv::Mat img;
        img = cv::imread(imgs_folder + "/" + img_names_vec[idx_frame].second, cv::IMREAD_GRAYSCALE);
        if(img.empty()) {
            std::cout << "img at " << t_img << " is empty!" << std::endl;
            continue;
        }
        if(img.channels() == 3) {
            cv::cvtColor(img, img, cv::COLOR_BGR2GRAY);
        }

        ov_core::CameraData measure_img;
        measure_img.timestamp = t_img;
        measure_img.sensor_ids.push_back(0);
        measure_img.images.push_back(img);

        Eigen::Matrix<double, 17, 1> imustate;
        if(!gt_states.empty() && !sys->initialized() && get_gt_state(measure_img.timestamp, imustate, gt_states)) {
            //biases are pretty bad normally, so zero them
            //imustate.block(11,0,6,1).setZero();
            sys->initialize_with_gt(imustate);
        } else if(gt_states.empty() || sys->initialized()) {
            sys->feed_measurement_camera(measure_img);
        }

        cv::Mat tracked_img = sys->get_historical_viz_image();
        if(!tracked_img.empty()) {
            double s = std::min(1.0, 1920.0 / std::max(tracked_img.rows, tracked_img.cols));
            if(s < 1.0) {
                cv::resize(tracked_img, tracked_img, cv::Size(), s, s);
            }
            cv::imshow("tracked_img", tracked_img);
        }
        int key = cv::waitKey(5);
        if(key == 'q' || key == 'Q') {
            break;
        }
    }
```

Do you think there is anything wrong with my code? @goldbattle

CanCanZeng, 2021-04-25:
Sorry, I made a misstake. The pose saved from rosbag is defined in imu corrdinate system, but the pose from raw images is defined in cam0 corrdinate system. When I transform the latter to imu coordinate system, the poses are exactly alike.

closed 2021-04-25 by CanCanZeng

goldbattle, 2021-06-09:
If you post your code, I can work to incorporate this directly into the repo since others might want to load it from the asl folder format. Thanks1

CanCanZeng, 2021-06-19:
Hi @goldbattle , I create a pull request for your reference.