rpng/open_vins #164: Decrease time delay/computation time of outputs.

TheRealKaufmann, 2021-05-12:
Hi there,

I am using OpenVins quite successfully on a real quadcopter. I am using the imu pose topic for that. 
But there is one problem: The data seems to have a delay of approx. 270ms, which is quite challenging to work with. 
I measured this by comparing the timestamps of the quadcopter attitude and the estimated attitude by OpenVins.

Is the computation time really that slow? I am using a Jetson Xavier NX on 4 Cores and OpenVins only "occupies" only 40% of CPU usage.

I hope you can help me.

Thanks,
Tom

WoosikLee2510, 2021-05-12:
The pose estimation should be processed faster than that time if the parameters are set at reasonable numbers. What's the estimated time offset between the IMU can the camera?

TheRealKaufmann, 2021-05-12:
Hi,

thanks for your answer. The time offset is around 0.00985. I am using the launch file for the T265 camera and it's IMU you provided in another issue I think (of course with own intrinsics and extrinsics).

WoosikLee2510, 2021-05-12:
If you are subscribing the topics, what's the size of the subscriber queue? 
Maybe the processing speed is slow so the image topics are piled up (meaning that the latest topic should wait to be processed). 
If the queue size is a large number, try setting it 0 or 1. This will drop the topics while processing measurement but allows you to process the latest topics. without delay.

goldbattle, 2021-05-12:
You can also try looking at the odometry topic timesteps as compared to the pose one as the odometry will try to use the IMU to propagate from the latest state time to the most recent IMU to facility low latency estimation. The pose_imu topic will still be the most accurate though.

How long is the update reported to take?
Have you tried to record a timing csv file and plotting it?https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_ros_eth.launch#L58-L60
https://docs.openvins.com/eval-timing.html#eval-ov-timing-flame

I would also make sure you are passing the correct optimization flags so that it optimizes for ARM.
Additionally, what camera resolution are you using? 640x480?

TheRealKaufmann, 2021-05-13:
The size of the subscriber queue is set to 1 and should not cause issues. I also tried grabbing the imu frame directly and converting it into a pose. 

I tried as @goldbattle suggested and ran tests with the ov_timing tool.
The first test was conducted by using the current setup/configs. The second test was conducted after building in release mode and setting ARM optimization flags. 

First test (using current config):
[image]

Second test (with optimization and release mode):
[image]

It does seem as it does not make a big difference. What exactly do you mean by ARM optimization flags? The flags I used specified the specific arm architecture I am using (armv8.2-a). Are there more useful flags?

The camera resolution 848x800 px. It is the standard resolution of the T255 camera.

Would it help, if I post my launch file?

Thanks!

WoosikLee2510, 2021-05-13:
T265 gives the image measurement 30 fps which means you need to process a measurement under 33 ms for real-time application. Even if you drop 2/3 of the measurements, you still have 100 ms which seems not feasible in your current setup.

I suggest you try tuning the parameters. 

Especially lowering `num_pts` can reduce tracking time,
```
<param name="num_pts"            type="int"    value="250" />
```

and reducing these numbers can reduce update times.
```
<param name="max_clones"             type="int"    value="11" />
<param name="max_slam"               type="int"    value="75" />
<param name="max_slam_in_update"     type="int"    value="25" /> 
<param name="max_msckf_in_update"    type="int"    value="40" />

```

additionally, try enabling the downsampling
```
<param name="downsample_cameras" type="bool"   value="false" />
```

TheRealKaufmann, 2021-05-13:
Thank you very much for your suggestions! I have tried to reduce them (even though they are quite low already), but that made the tracking sometimes unreliable. Also, I noticed when using the downsampling that the execution time is significantly higher (about 100ms higher). 

My current setup is: 
`<param name="num_pts"            type="int"    value="150" />
<param name="max_clones"             type="int"    value="11" />
<param name="max_slam"               type="int"    value="25" />
<param name="max_slam_in_update"     type="int"    value="25" />
<param name="max_msckf_in_update"    type="int"    value="50" />`

I still don't understand exactly what is holding me back from running this in realtime or at least with a <100ms delay? What kind of setup changes would be required for that?

Thanks!

goldbattle, 2021-05-14:
We have been able to out of the box run on the Jetson Nano, [see this work](https://udel.edu/~pgeneva/downloads/papers/c19.pdf), at 28-30fps so this seems a bit weird that your tracking is taking >0.03 seconds. I recommend you try either increasing the FAST threshold so it detects less features, and also try re-compiling OpenCV from scratch to see if it is OpenCV not using the correct optimization flags.

I believe passing the correct ARM arch and using `-O3` is really all that is needed in general.

TheRealKaufmann, 2021-05-18:
I have recompiled OpenVins and OpenCV with -O3 and -march optimization flags and it got a little better but still not great.

[image]

One more question: Does OpenVins prefer multiple CPU cores to work with over a single one or two? When I limit my system to 2 cores, I can get a better CPU maximal frequency compared to when I use 4 cores for example.

Thanks!

goldbattle, 2021-05-29:
Right now there is no GPU acceleration code.  Can you post an example ROS bag along with your launch file that you are using? I hope that we can work out the issue since I have never run into such large times on our TX2 / Nano / Raspi.

TheRealKaufmann, 2021-05-31:
Hi, thank you very much for your help. 
Here is the link with the launch file and bag:
https://drive.google.com/drive/folders/18rPvuktzhf8pvnCALhRnUKGllr_asult?usp=sharing

TheRealKaufmann, 2021-06-01:
Update: I have tried the same thing on a Jetson TX2 without any optimizations. The result seems better than on the Xavier NX.
[image]

goldbattle, 2021-06-21:
Your launch file looks reasonable to me. You can also try enabling down sampling and single threaded mode with the higher core frequency (see this launch file that I edited):
https://gist.github.com/goldbattle/e8c5bc2baf7fd1b2eb0c2f0e79f2aa36

Also, I am just now realizing that you are running stereo. I believe all the evaluation and timings I have done have been for a monocular system. Can you try running binocular or monocular and see how that performs vs stereo?

TheRealKaufmann, 2021-06-30:
Yes I will try that. Since it seems like more of an platform-specific issue, I will close this for now. Thank you very much for your help!

closed 2021-06-30 by TheRealKaufmann

bryangd34, 2024-08-20:
Hi @TheRealKaufmann 
Could you please share how you set the optimization flags (`-O3` and `-march`). I can't find any documentation for how to add these flags to `colcon build` . Thanks.