rpng/open_vins #169: Could you sharing the detail information about online intrinsic camera calibration

arenas7307979, 2021-05-30:
Thanks for sharing! That's an awesome work
Could you  introduce the cost function and Jacobin of camera intrinsic parameters?

goldbattle, 2021-05-30:
I recommend you read the paper and documentation.
http://udel.edu/~pgeneva/downloads/papers/c10.pdf
https://docs.openvins.com/update-feat.html#distortion

arenas7307979, 2021-06-08:
Thank you for your reply.

arenas7307979, 2021-06-24:
hii, 
I found a difference between the document and  code, 
code :  (2 * cam_d(4) * x_2 + 4 * cam_d(5) * x_2 * r)
document: 2 * p1 * xn^2 + 4 * k2  * xn^2  *（xn^2 + yn^2) 
Should I change **r** to **r squared**？

[image]

[image]

[image]

goldbattle, 2021-07-06:
Thanks, documentation is correct. Will correct in an upcoming commit.

PR #186 (2021-07-19): Development v2.4 - Camera classes, Masks, Formatting and Smaller Fixes
## Key Changes
- Camera class objects have been added which consolidate both the distortion / undistortion / Jacobian functionality
- The frontend has been reworked to take in this new camera object and the "CameraData" message objects directly
- Support for masked areas of the image has been added to the tracking (closes #159)
- Docker use and documentation has been added (originally mentioned in #167)
- Fix radtan norm intinsic jacob (fixes #169)
- Add live alignment of groundtruth if we have it (for better dev'ing)
- Dat [... cut]
commit 9f4d3e9b (2021-07-19): Fix radtan norm intinsic jacob (fixes #169)
closed 2021-07-19 by goldbattle