rpng/open_vins #171: Consistency maintenance methods, FEJ vs Observability-constrained(OC) ones

lewisjiang, 2021-06-04:
Dear authors,

Thank you for your wonderful platform suite. 

I was reading some papers about keeping VINS system consistent. In [Huang IJRR10](http://journals.sagepub.com/doi/10.1177/0278364909353640), [Hesch IJRR14](http://journals.sagepub.com/doi/10.1177/0278364913509675), two different observability-constrained methods are mentioned apart from FEJ, in order to maintain consistency. And in [Huang IJRR10](http://journals.sagepub.com/doi/10.1177/0278364909353640), the authors find OC-EKF performs slightly better than FEJ, since the linearization error is smaller. 

It seems that only FEJ is implemented in OpenVINS. As far as I know, [S-MSCKF](https://github.com/KumarRobotics/msckf_vio) uses Hesch's OC method. May I ask why the observability-constrained methods are not used in OpenVINS?

Thanks,
Lewis

goldbattle, 2021-06-09:
I recommend you read Section 5 of:
https://journals.sagepub.com/doi/10.1177/0278364909353640

The papers do a good job comparing and contrasting each method, thus I recommend you refer those as a primary source of information. I think one point is that one has to find the nullspace / an unobservable subspace to pick when using OC, while with FEJ one can blindly apply it and handle all directions. For example OC would be unable to handle constant acceleration or pure translation degeneracy while if we use FEJ it should already be able to be handled properly.

Some other thoughts for you to think about:
- FEJ is more simple to implement.
- OC smaller linearization errors.
- OC enforces a *specific* observable direction which might not be the best or optimal choice (many different U matrices could be chosen).

Remember that the goal of OpenVINS is to allow the community to extend the codebase, which the FEJ method's simplicity greatly helps with ensuring (just requires a small change to the Jacobian code as you can see in the updaters).

lewisjiang, 2021-06-15:
> I recommend you read Section 5 of:
> https://journals.sagepub.com/doi/10.1177/0278364909353640
>
> The papers do a good job comparing and contrasting each method, thus I recommend you refer those as a [...]

Thank you for your reply, I think I need to look at the OCs more carefully.

closed 2021-06-15 by lewisjiang