rpng/open_vins #172: Monocular VIO divergent behavior

gchustz, 2021-06-09:
Hello,

First, thank you for the amazing body of work you have graciously provided!

I have been trying to perform monocular VIO using a Vectornav VN300 IMU and a basler Pylon using only MSCKF features and IMU propagation. Meaning I have all SLAM feature limits set to 0. There seems to be some major discrepancy between the camera and the IMU as the runs are very divergent, jumpy, etc. even though I am getting good KLT tracks.

When running with IMU propagation only at kalibr-allan acquired values, the motion is smooth and matches with what occurring until IMU drift occurs; however, when I allow MSCKF features the tracking quickly diverged and runs away. 

With low chi2 (1-2) and sigma px (1-2) values it let's basically no features pass and has the divergent and jumpy behavior.

With high chi2 (25-30) and sigma px (3-5) it finally allows MSCKF features but the tracking is still horrible. It's only when I increase the IMU accelerometer by >100x does it seem to faithfully use the MSCKF features but the scale and motion is wrong. https://github.com/rpng/open_vins/issues/102#issuecomment-702243380 https://github.com/rpng/open_vins/issues/147#issuecomment-793107432

Can you take a look at my launch file and bag to let me know if I'm doing anything wrong? I have an inkling that my camera intrinsics are off and need to be recalibrated because of the decent performance with just the IMU.

I also wanted to ask for clarification about the IMU initialization threshold: should it be right above the static noise such that the moment it moves the threshold is exceeded, or should the threshold be high and one should perform a large excitation at the beginning of the run? https://github.com/rpng/open_vins/issues/93#issuecomment-678667821

Any help in properly implementing your work is greatly appreciated!

`<launch>

    
    <arg name="max_cameras" default="1" />
    <arg name="use_stereo"  default="false" />
    <arg name="bag_start"   default="5" /> 
    <arg name="bag"         default="/<path_to>/lab_traverse_medium_trimmed_2021-06-02.bag" />

    
    <arg name="init_window_time"  default="1.0" />
    <arg name="init_imu_thresh"   default="3.5" />

    
    <arg name="dosave"      default="false" />
    <arg name="dotime"      default="false" />
    <arg name="path_est"    default="/tmp/traj_estimate.txt" />
    <arg name="path_time"   default="/tmp/traj_timing.txt" />

    
    <node name="run_subscribe_msckf" pkg="ov_msckf" type="run_subscribe_msckf" output="screen" clear_params="true" required="true">

        
        <param name="topic_imu"      type="string" value="/vectornav/IMU" />
        <param name="topic_camera0"  type="string" value="/pylon_camera_node/image_raw" />

        
        <param name="use_fej"                type="bool"   value="true" />
        <param name="use_imuavg"             type="bool"   value="true" />
        <param name="use_rk4int"             type="bool"   value="true" />
        <param name="use_stereo"             type="bool"   value="$(arg use_stereo)" />
        <param name="calib_cam_extrinsics"   type="bool"   value="false" />
        <param name="calib_cam_intrinsics"   type="bool"   value="false" />
        <param name="calib_cam_timeoffset"   type="bool"   value="true" />
        <param name="calib_camimu_dt"        type="double" value="0.0" />
        <param name="max_clones"             type="int"    value="11" />
        <param name="max_slam"               type="int"    value="0" />
        <param name="max_slam_in_update"     type="int"    value="0" /> 
        <param name="max_msckf_in_update"    type="int"    value="40" />
        <param name="max_cameras"            type="int"    value="$(arg max_cameras)" />
        <param name="dt_slam_delay"          type="double" value="3" />
        <param name="init_window_time"       type="double" value="$(arg init_window_time)" />
        <param name="init_imu_thresh"        type="double" value="$(arg init_imu_thresh)" />
        <rosparam param="gravity">[0.0,0.0,9.81]</rosparam>
        <param name="feat_rep_msckf"         type="string" value="GLOBAL_3D" />
        <param name="feat_rep_slam"          type="string" value="ANCHORED_FULL_INVERSE_DEPTH" />
        <param name="feat_rep_aruco"         type="string" value="ANCHORED_FULL_INVERSE_DEPTH" />

        
        <param name="try_zupt"               type="bool"   value="false" />
        <param name="zupt_chi2_multipler"    type="int"    value="2" />
        <param name="zupt_max_velocity"      type="double" value="0.3" />
        <param name="zupt_noise_multiplier"  type="double" value="50" />

        
        <param name="record_timing_information"   type="bool"   value="$(arg dotime)" />
        <param name="record_timing_filepath"      type="string" value="$(arg path_time)" />

        
        <param name="use_klt"            type="bool"   value="true" />
        <param name="num_pts"            type="int"    value="400" />
        <param name="fast_threshold"     type="int"    value="10" />
        <param name="grid_x"             type="int"    value="5" />
        <param name="grid_y"             type="int"    value="3" />
        <param name="min_px_dist"        type="int"    value="5" />
        <param name="knn_ratio"          type="double" value="0.85" />
        <param name="downsample_cameras" type="bool"   value="false" />
        <param name="multi_threading"    type="bool"   value="true" />

        
        <param name="use_aruco"        type="bool"   value="false" />
        <param name="num_aruco"        type="int"    value="1024" />
        <param name="downsize_aruco"   type="bool"   value="true" />

        
        <param name="up_msckf_sigma_px"            type="double"   value="2" />
        <param name="up_msckf_chi2_multipler"      type="double"   value="30" />
        <param name="up_slam_sigma_px"             type="double"   value="2" />
        <param name="up_slam_chi2_multipler"       type="double"   value="30" />
        <param name="up_aruco_sigma_px"            type="double"   value="1" />
        <param name="up_aruco_chi2_multipler"      type="double"   value="1" />
        
        <param name="gyroscope_noise_density"      type="double"   value="8e-05" /> 
        <param name="gyroscope_random_walk"        type="double"   value="5e-06" />
        <param name="accelerometer_noise_density"  type="double"   value="8e-02" />
        <param name="accelerometer_random_walk"    type="double"   value="8e-04" />

        
        <rosparam param="cam0_wh">[1920, 1200]</rosparam>
        <param name="cam0_is_fisheye" type="bool" value="false" />
        <rosparam param="cam0_k">[1060.7286, 1059.8339,966.9318, 584.0839]</rosparam>
        <rosparam param="cam0_d">[-0.15463337, 0.10234123, 0.00018375, 0.00048037]</rosparam>

        
        <rosparam param="T_C0toI">
            [
            -0.010051784125709257, 0.9999493504081445, -0.0005081871924421111, -0.1680434499347249,
            -0.008999339197121481, 0.00041772831201095095, 0.999959417875083, 0.05558435723294967,
            0.9999089826228724, 0.010055949551870613, 0.00899468446944865, -0.1275121065599258,
            0.0, 0.0, 0.0, 1.0
            ]
        </rosparam>

    </node>

    
    <node pkg="rosbag" type="play" name="rosbag" args="-d 1 -s $(arg bag_start) $(arg bag)" required="true"/>

    
    <group if="$(arg dosave)">
        <node name="recorder_estimate" pkg="ov_eval" type="pose_to_file" output="screen" required="true">
            <param name="topic"      type="str" value="/ov_msckf/poseimu" />
            <param name="topic_type" type="str" value="PoseWithCovarianceStamped" />
            <param name="output"     type="str" value="$(arg path_est)" />
        </node>
    </group>

    <node type="rviz" name="rviz" pkg="rviz" args="-d $(find ov_msckf)/launch/display.rviz" />
</launch>
`
bag file : https://drive.google.com/drive/folders/1o3ATVN4JE5v2xazPDtoY5SC7dSiC9Few?usp=sharing

goldbattle, 2021-06-09:
Your IMU noises are probably too small if you just directly took them. You probably need to inflate by 1-2 orders. Additionally, your extrinsics look like it is 16 and 12 centimeters in the x-z directions which seem a bit large. Does this match hand measured values? I recommend you take a look and run with the existing supported datasets first to get a feel and look a the [calibration guide](https://docs.openvins.com/gs-calibration.html) which might help you.

gchustz, 2021-06-09:
@goldbattle Thank you for your rapid response, 

I will let you know my findings, hopefully later today.

gchustz, 2021-06-16:
@goldbattle Thank you so much for bringing up the timing issues with vectornav, because after a weekend of modifying drivers we have corrected the vectornav timestamps and also gotten the pylon camera outputting timestamps synced with PTP. The calibration with kalibr and the OV MSCKF tracking is much, **much** better and stable. I am still working on some tracking issues with respect to certain motion and losing MSCKF features, but I have yet to spend time on tuning.

Attaching my launch file for transparency and for further suggestions:

```
<launch>

    <node name="run_serial_msckf" pkg="ov_msckf" type="run_serial_msckf" output="screen">
    
        <!-- bag topics -->
        <param name="topic_imu"      type="string" value="/vectornav/IMU" />
        <param name="topic_camera0"  type="string" value="/pylon_camera_node/image_raw" />
    
        <!-- bag parameters -->
        <param name="path_bag"    type="string" value="/<path_to>/aruco_cross1_2021-06-16-13-43-15.bag" />
        <param name="bag_start"   type="int"    value="0" />
        <param name="bag_durr"    type="int"    value="-1" />
    
        <!-- world/filter parameters -->
        <param name="max_clones"             type="int"    value="11" />
        <param name="max_slam"               type="int"    value="0" />
        <param name="max_cameras"            type="int"    value="1" />
        <param name="init_window_time"       type="double" value="0.75" />
        <param name="init_imu_thresh"        type="double" value="0.10" />
        <rosparam param="gravity">[0.0,0.0,9.81]</rosparam>
        <param name="feat_rep_msckf"         type="string" value="GLOBAL_3D" />
        <param name="feat_rep_slam"          type="string" value="GLOBAL_3D" />
    
        <!-- tracker/extractor parameters -->
        <param name="use_klt"          type="bool"   value="true" />
        <param name="fast_threshold"   type="int"    value="10" />
        <param name="grid_x"           type="int"    value="5" />
        <param name="grid_y"           type="int"    value="3" />
        <param name="min_px_dist"      type="int"    value="10" />
        <param name="num_pts"          type="int"    value="800" />
    
        <!-- sensor noise values / update -->
        <param name="up_msckf_sigma_px"            type="double"   value="1" />
        <param name="up_msckf_chi2_multipler"      type="double"   value="1" />
        <!-- x10 values given -->
        <param name="gyroscope_noise_density"      type="double"   value="6.2e-03" />
        <param name="gyroscope_random_walk"        type="double"   value="5.0e-04" />
        <param name="accelerometer_noise_density"  type="double"   value="1.051e-01" />
        <param name="accelerometer_random_walk"    type="double"   value="3.2e-03" />
    
        <!-- camera intrinsics -->
        <param name="cam0_is_fisheye" type="bool" value="false" />
        <rosparam param="cam0_k">[1053.8767420505942, 1052.6765254287882, 967.3899880574555, 587.1422341787483]</rosparam>
        <rosparam param="cam0_d">[-0.1374906235082657, 0.05783702444800811, 0.000180013561902521,
    0.0006508618187655724]</rosparam>
    
        <!-- camera extrinsics -->
        <rosparam param="T_C0toI">
            [
            -0.01225928, -0.02432092,  0.99962903,  0.07620639,
             0.9999158,   0.00395521,  0.01235903,  0.13638911,
            -0.00425433,  0.99969638, 0.02427039, -0.06510433,
            0.0, 0.0, 0.0, 1.0
            ]
        </rosparam>
    
    </node>
    
    <node type="rviz" name="rviz" pkg="rviz" args="-d $(find ov_msckf)/launch/display.rviz" />
    
</launch>
```
[image]

closed 2021-06-16 by gchustz