rpng/open_vins #176: Camera Extrinsic

Hughbyun, 2021-06-18:
Hi,
 
I am just wondering about the camera extrinsic. 

For the parameter T_C0toI, I put the transformation matrix from the camera frame to IMU. 
And I also tried with the parameters from VINS-fusion, which is from IMU to the camera frame.  
However, it shows better pose estimation when I use the param from VINS which is also showing differences in feature detection as can be seen in the pictures below. I am using the same value except camera extrinsic. 

- T_C0toI
[image]

- body_T_cam0
[image]

Which one is the right camera extrinsic to put for OpenVINS?
Looking forward to hearing from you. 
Thanks.

goldbattle, 2021-06-18:
OpenVINS uses T_C0toI which means it contains the rotation from CAM0 to IMU (R_C0toI) and the position of the CAM0 in the IMU frame (p_C0inI). I believe we use the same transformation as VINS-Mono (if my memory is correct, you can compare the eth/euroc configs). So you might have misunderstood the variable naming of VINS-Mono which is a bit confusing.

For ours see this:
https://docs.openvins.com/dev-coding-style.html#coding-style-naming

Hughbyun, 2021-06-18:
Thanks for the feedback. 
Then I am sure I misunderstood the variable naming. 
I should have compared the configs, sorry for bothering you. 
Thanks again.

closed 2021-06-18 by Hughbyun

goldbattle, 2021-06-18:
No worries, glad to help. You can always try disabling online calibration as the system should still run with reasonable values, but if you flip the extrinsic by accident then it will normally diverge / fly away instantly.

Hughbyun, 2021-06-22:
Hi @goldbattle, 

Sorry to bother you again. 
Just wondering about the Schmidt-MSCKF. Are you planning to open-source for S-MSCKF, too?
Thanks. Looking forward to hearing from you.

goldbattle, 2021-06-22:
Not at this time, down the line maybe.