rpng/open_vins #184: support camera model

NightNightNight, 2021-07-08:
Does open_vins support panoramic camera like Ricoh or insta360 or not?

goldbattle, 2021-07-08:
As per https://github.com/rpng/open_vins/issues/131 we still don't. I believe the main issue is that these cameras are no longer pinhole cameras, thus need a different triangulation and projection models. It might be something we should support in the future if there is enough interest.

Do you have links to what drivers / exact cameras you were planning on using?
Do they come with IMU, or do you need to attach it as an additional sensor?

NightNightNight, 2021-07-09:
https://www.insta360.com/cn/product/insta360-onex/

NightNightNight, 2021-07-09:
we used Ricoh camera without IMU. Now we will use insta360 with IMU.  [https://www.insta360.com/cn/product/insta360-onex/](url)

CanCanZeng, 2021-07-22:
Hi @NightNightNight , how do you extract imu data from insv file produced by insta360 device?
As I have tested, gopro max is better to do vio, since it has done much things to remove rolling shutter effect and stitch effect. And they provide an open source project https://github.com/gopro/gpmf-parser to extract meta data(imu and gps) from their video. Insta360 produces more visual appealing video, but is seems not very good for geometrical machine vision task.

goldbattle, 2022-03-21:
Closing this since it isn't something that we plan to add at this point (someone can fork and implement if they are interested).

closed 2022-03-21 by goldbattle

XUTONG1024, 2024-07-04:
> Does open_vins support panoramic camera like Ricoh or insta360 or not?

May I ask if you have made any progress on this issue? I am currently considering using the built-in IMU of Insta360 x2 to implement vio. Can you give me some suggestions?

XUTONG1024, 2024-07-04:
> Hi @NightNightNight , how do you extract imu data from insv file produced by insta360 device? As I have tested, gopro max is better to do vio, since it has done much things to remove rolling shutter [...]

May I ask if you have made any progress on this issue? I am currently considering using the built-in IMU of Insta360 x2 to implement vio. Can you give me some suggestions?