rpng/open_vins #185: Process [rosbag-2]has died'

chenyong1003, 2021-07-13:


chenyong1003, 2021-07-13:
[ZUPT]: failed disparity (1.071 > 0.500, 332 features)
[ZUPT]: rejected |v_IinG| = 0.010 (chi2 170.722 > 0.000)
[TIME]: 0.0100 seconds for tracking
[TIME]: 0.0009 seconds for propagation
[TIME]: 0.0001 seconds for MSCKF update (0 feats)
[TIME]: 0.0076 seconds for SLAM update (65 feats)
[TIME]: 0.0008 seconds for SLAM delayed init (0 feats)
[TIME]: 0.0025 seconds for re-tri & marg (11 clones in state)
[TIME]: 0.0218 seconds for total (camera 0 1)
q_GtoI = 0.816,-0.053,0.574,0.045 | p_IinG = -0.349,0.000,0.090 | dist = 78.39 (meters)
bg = -0.0015,0.0152,0.0785 | ba = -0.0178,0.1781,-0.0022
camera-imu timeoffset = -0.00020
cam0 intrinsics = 458.520,458.541,364.830,246.228 | -0.277,0.068,0.000,-0.000
cam1 intrinsics = 456.793,456.730,377.821,252.999 | -0.277,0.069,-0.000,0.000
cam0 extrinsics = -0.005,0.009,0.702,0.712 | 0.069,-0.011,0.008
cam1 extrinsics = -0.000,0.014,0.702,0.712 | -0.041,-0.009,0.003
[TIME]: 0.0016 seconds for visualization
================================================================================REQUIRED process [rosbag-2] has died!
process has finished cleanly
log file: /home/c/.ros/log/fb4126b6-e3d3-11eb-8efc-000c2941d447/rosbag-2*.log
Initiating shutdown!
================================================================================
[rosbag-2] killing on exit
[run_subscribe_msckf-1] killing on exit
camera-imu timeoffset = -0.00020

cam0 intrinsics:
458.520,458.541,364.830,246.228
-0.27682,0.06801,0.00013,-0.00002

cam1 intrinsics:
456.793,456.730,377.821,252.999
-0.27749,0.06897,-0.00019,0.00004

T_C0toI:
0.015,-1.000,0.006,-0.012,
1.000,0.015,0.020,-0.069,
-0.020,0.006,1.000,-0.007,
0.000,0.000,0.000,1.000

T_C1toI:
0.013,-1.000,0.019,-0.008,
1.000,0.013,0.020,0.041,
-0.020,0.019,1.000,-0.004,
0.000,0.000,0.000,1.000

TIME: 108.892 seconds

terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
shutting down processing monitor...
... shutting down processing monitor complete
done
The above error occurred in the running .can you tell me what the problem is？Thank you very much！

goldbattle, 2021-07-13:
It looks like it ran fine, see that the process finished cleanly.

chenyong1003, 2021-07-14:
Thank you for response！I use the launch file below: https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/pgeneva_ros_eth.launch. The above description appears when it runs,and the corresponding file isn’t generated in the corresponding folder.I reinstalled all the software,but nothing worked out.

goldbattle, 2021-07-14:
What corresponding file? You should be able to see everything in RVIZ, use the config file in the ov_msckf/launch/ folder for that.

chenyong1003, 2021-07-14:
When the program finished,the save track path wasn’t generated in the specified folder tmp.

goldbattle, 2021-07-14:
I recommend you check permissions and that the folder is already created.
Are there any error logs from the `pose_to_file` in the console?
Are you able to visualize everything in RVIZ ok?
I assume you have enabled the param to record to file correct?

chenyong1003, 2021-07-15:
Can you elaborate on the error log？After I reinstalled the system,RVIZ wasn’t displayed.Then I added it according to the original one, and now it can be displayed. It wiil be the same after I finish running, [run_subscribe_msckf-1]killing on exit and [rosbag-2]killing on exit will be displayed on the terminal.

goldbattle, 2021-07-15:
There was no error in what you posted, can you provide a bit more detail on the issue?

chenyong1003, 2021-07-16:
[image]
[image]
[image]
This is the picture taken after I just finished running. Is there anything else to add？

goldbattle, 2021-07-16:
Have you set the dosave to true? You can also try saving to another folder.

chenyong1003, 2021-07-18:
Thank you very much for you reply！now the track can be saved,but the terminal still shows [rosbag-2]and[run_subscribe_msckf-1]killing on exit.

goldbattle, 2021-07-19:
Glad we got it working. The rosbag will terminate once it finishes playing the complete bag.

closed 2021-07-19 by goldbattle