rpng/open_vins #187: Gravity in the global frame

rakshith95, 2021-07-14:
Hello, does the 'gravity' parameter depend on the axes of the IMU, or only the 'world' frame? The documentation says "Gravity in the global frame (i.e. should be [0, 0, 9.81] typically)"; and these are my world, and IMU frames.  [image] 

Since the world frame has the gravity along the Z axis, I assume it must be defined as [0,0,9.81], but I wonder if it should differ since my IMU has the -Y axis

WoosikLee2510, 2021-07-14:
Hi. Are you using IMU with NED coordinate? In that case, I saw no problem blindly using the system as it is, except that the trajectory was flipped.

rakshith95, 2021-07-14:
Hi, the IMU does not follow the NED coordinate system.  It is similar to the OpenCV system where the Z axis protrudes out, + Y axis is towards the ground, and X axis  to the right

WoosikLee2510, 2021-07-14:
If it is just how you posed the imu, then it should be fine.

rakshith95, 2021-07-14:
Alright, thank you. Can you tell me when it would be different?

WoosikLee2510, 2021-07-14:
I hope my previous (edited) comment helps?

rakshith95, 2021-07-15:
Yes it does, thank you. Closing this.

closed 2021-07-15 by rakshith95