rpng/open_vins #188: trajectory problem

chenyong1003, 2021-07-26:
sorry to bother you again! I can't use EVO to evaluate my data after it is saved.Is there any way to solve this?
[image]

goldbattle, 2021-07-26:
We recommend you use the built in ov_eval utility instead if you need an out of the box solution.
https://docs.openvins.com/namespaceov__eval.html

Otherwise you will need to convert to their format if needed, note that we record with a space separated file.
Referred to as the RPE format instead of the Euroc file format and vice-versa.
See the documentation site:
https://docs.openvins.com/eval-error.html#eval-ov-transform

I am not sure if the [EVO](https://github.com/MichaelGrupp/evo) package supports space separated, but you could try looking at their documentation or opening an issue on their github. Hope this helps.

chenyong1003, 2021-07-27:
I used the method you said,it appeared the following problem.
[image]

goldbattle, 2021-07-27:
You need to source your ros workspace once you build the open_vins ov_eval package.

closed 2021-07-29 by goldbattle