rpng/open_vins #189: Questions about the test environment

changseung, 2021-07-28:
I have a question.
I'm going to do an experiment with a drone and I'm going to install a net in four directions for safety reasons.
Do you think the dense net or pattern will affect the performance of openvins?
Dense green safety net like this. https://www.wikiwand.com/en/Safety_net

goldbattle, 2021-07-29:
Should be fine if it is able to reliably track features.
You can also attach aruco tags to the walls / floor if you want more stable estimation for indoors.

changseung, 2021-07-30:
Thank you for your answer.

Can i ask you one more question?
In fact, I tested it using the net and found the same phenomenon as the video below. I think this is a feature matching failure due to the dense net. What do you think?

https://user-images.githubusercontent.com/78731953/127624380-41cc6133-a2cb-4a29-b125-8d48af9cce6d.mp4

goldbattle, 2021-08-07:
I think your observation is correct there. The nets we used in the pass are much larger hole size (around 3-4in squares) so we had less issues I think. You can also try facing the camera down more at a 45 degree. Not sure if it is possible to get around this issue currently.

changseung, 2021-08-19:
Thank you for your advice.

closed 2021-08-19 by changseung