rpng/open_vins #190: Memory Usage increase rapidly

Majed-Alsubaie, 2021-07-31:
Hi,,
Thank you for releasing such a project ,, it is wonderful 

I face an issue with the memory where i do small movement  and stop then " [ZUPT]: chi2_check over the residual limit - 1878" appears on the terminal
After a while memory usage start jump!

see the video:https://drive.google.com/file/d/1xwCiOIDcFVQd1x-HqOusPo-TWpSr_R8z/view?usp=sharing

i have tested the same launch file with Mh-01 ,,, the algorithm did great! 
the launch file : https://drive.google.com/file/d/11EJbk5BCRNWUasFv4mxewFNC9Bn5nGgm/view?usp=sharing

goldbattle, 2021-08-03:
This might be due to how we keep a history of visual tracks and IMU readings.
Are you using the latest commit from the main branch?

goldbattle, 2021-08-23:
This also seems to be reported in this paper, needs to be investigated.
https://arxiv.org/pdf/2108.01654.pdf
[image]

Probably due to how we store IMU data and need to have better logic to handle that deletion or how we store historical image track information in our feature database (maybe measurement delete logic doesn't work?).

goldbattle, 2022-01-06:
Unable to reproduce currently on my machine (Ubuntu 20.04).

closed 2022-01-06 by goldbattle