rpng/open_vins #191: Dev-roadmap question

brunopinto900, 2021-08-03:
First of all, congratutions for such amazing work!

In this link https://docs.openvins.com/dev-roadmap.html , i noticed that you intend to implement "Large scale offline batch graph optimization which leverages the trajectory of the ov_msckf as its initial guess and then optimizes both the map and trajectory."

Is this already done? In progress? 

I am using MSCKF-based VIO for the Formula Student Germany competion as part of a team. I was thinking of adding loop closure, factor graphs for global map-trajectory optimization. 

Thank you.

goldbattle, 2021-08-03:
No timeline since this is a research project, thus no promises for any sort of delivery, but it is something we are working on internally. You might also want to checkout the [ov_secondary](https://github.com/rpng/ov_secondary) project along with what we publish for secondary loop-closure algorithms to use: https://github.com/rpng/open_vins/blob/master/ov_msckf/src/core/RosVisualizer.cpp#L679-L841

brunopinto900, 2021-08-03:
Yes, of course. Good luck. I didn't understand though, are you working on adding models like for example the car model (simple bicycle dynamics)? If not, its not an issue, i want to do it for the competition.

goldbattle, 2021-08-03:
Ah yes, sorry I missed the last part. You could add models possibly, but it would depend on how you plan to incorporate those vehicle dynamics into the system. You could design your own measurement and update the state or change how the state evolves in the propagation stage.

closed 2021-08-07 by goldbattle